witnessmenow/Universal-Arduino-Telegram-Bot #3: Library should throw exceptions when if it runs into issues

witnessmenow, 2016-07-05:
Currently the library writes to the serial if it runs into an issue, but it should be throwing exceptions

jandrioli, 2018-01-25:
You may close this issue, since on AVR architectures there is no way to handle exceptions. And probably there never will be.

witnessmenow, 2018-01-26:
Good to know, I might leave this open as a task to improve handling of non sunny paths, maybe we could return error codes etc

jandrioli, 2018-01-27:
What's the plan for integrating those PRs ?
They work, in fact I have improvements to push.

witnessmenow, 2018-01-27:
The PR needs some modification before approving and I just haven't gotten
around to it yet.

If you want to raise a PR with your changes, if it contains both, I could
use that as the starting point

Thanks