witnessmenow/Universal-Arduino-Telegram-Bot #5: Improve examples

witnessmenow, 2016-07-14:
The examples do not show the best way of using the library and need to be improved

witnessmenow, 2016-07-14:
@carlosjosetam let me know that he needed to make a change to the example to get it to connect to his wifi. See changes around Wifi.mode in the attached photo 

[image]

rusboard, 2017-01-08:
I think that WIFI_STA it's custom for this user.
All my sketches working fine without any WiFi.mode() and telegram working good.

I think that this Wifi.mode must be in code as comment with description, try uncomment this line if your Wifi not working (or something).

IMHO

witnessmenow, 2017-01-09:
Yeah, I know i don't use it in most of my own sketches anyways, but I rarely have my ESPs doing anything but being a regular client. 

I think it's ok to leave it in though as it doesn't seem to have a negative impact. For me the main purpose of the examples is to show off the Telegram functionality, I'm guess most users will have their own preferences for how they handle connecting to the WiFi etc (or at least that was my thinking!)

closed 2017-01-09 by witnessmenow