witnessmenow/Universal-Arduino-Telegram-Bot #7: Unable to send messages

petit-miner, 2016-09-12:
I tested the FlashledBot example, I changed the SSID and the password.
The ESP8266 connects sucessfully and can receive all of my messages via telegram, but it wont respond. I added a simple Serial.println to print the variable "text" to confirm that my messages arrive.

Furthermore I had a problem with the ArduinoJSON Lib.
The Arduino IDE couldn't find the ArduinoJSON lib, I checked the installation path twice, but still the IDE couldn't find the lib. To solve this I had to copy the complete ArduinoJSON lib to the same directory as the FlashledBot example.
Maybe a bad link or something like that?

witnessmenow, 2016-09-18:
Hey @petit-miner , sorry for the delay in getting back to you! New born babies reduce your free time considerably!

So i just tested the most up to date echo bot there and it's working fine for me. Can you describe your Telegram set up, are you talking directly to a bot or did you add the bot to a group?

Can you check the boolean return of sendMessage? it should return true if it thinks the message sent.

Feel free to message me on telegram, i'd probably be able to get back to you quicker there (@witnessmenow)

Not sure on the arduinoJson one, i think i installed it by just searching for it on the library manager, but im really not sure. I dont need to have the lib in the same folder for sure

witnessmenow, 2016-11-06:
Going to close as its working for me, let me know if its still an issue and i'll reopen

closed 2016-11-06 by witnessmenow