witnessmenow/Universal-Arduino-Telegram-Bot #8: Sending Images

Melkor82, 2016-12-11:
I was wondering to use arducam and send pictures through telegram bot to implement a very simple home monitoring system #

witnessmenow, 2016-12-11:
Hey melkor82,

Good idea! In theory it should be possible, and it is something I would like to add but I do not have an arducam though.

Maybe you could send me an image taken from an arducam and I can try get image sending working?

Thanks,
Brian

Melkor82, 2016-12-11:
Unfortunately in do not own a arducam too since i was planning to buy only if i could use it. But maybe you can directly ask them an image as sample, They are in github too : 
https://github.com/ArduCAM/ArduCAM_ESP8266_UNO
They encode in a standard jpeg format, so maybe if you succeed in sending a standard jpeg image (maybe stored in a sd) then he trick is done since they can save the images in sd cards

witnessmenow, 2016-12-13:
Did sime work on this tonight, was having some issues with the watch dog restarting. Will keep looking into it. I think i need to feed the watch dog while im uploading the image

Uploading an image from SD and direct from the Arducam will need to be different methods as one will pass in a file reference, and the other will pass in a reference to the arducam, so I don't think I'll be able to test it working without an actual arducam

witnessmenow, 2017-01-07:
My WIP on this so far

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/feature_sendImage

I don't think the format of my [message to Telegram](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/feature_sendImage/src/UniversalTelegramBot.cpp#L128) is correct so I might try send it via some way that works like curl and wireshark it to see what the message to Telegram is and try work backwards from there

rusboard, 2017-01-08:
I am working too that send images. But no luck right now. Trying, trying.

witnessmenow, 2017-01-09:
More info on my approach so far here

https://www.reddit.com/r/arduino/comments/5mp95m/trying_to_send_a_jpg_from_an_sd_card_via/

I have used the posttestserver as suggested and it also did not work for me so there is something wrong with my request.

I have updated some of what i was working on the feature branch if it helps anyone.

witnessmenow, 2017-01-09:
FINALLY!

[image]

It may not look like much, its just random picture i made that was small but there is an image sent via a Esp8266 via telegram

Now i just need to make the code usable!

rusboard, 2017-01-09:
Great job, man!

witnessmenow, 2017-01-10:
The image i sent was tiny (1k), I tried with a more normal image there and it was incredibly slow. Adding the bytes to a buffer and writing them when the buffer is full is much quicker, but its still pretty slow. A 2.5Mb image took around 10 minutes to send.

Maybe the buffer could be increased to speed it up (currently at 512, setting to 2048 caused a crash).

I'll check in what I have to the feature branch, but it still needs some more work to be merged to the master branch.

I also would like to avoid having a reference to the SD module in the library. A generic approach where it didnt matter the source of the image would be best if its possible.

I was thinking of maybe passing in methods calls , like [here](https://github.com/witnessmenow/esp8266-alexa-wemo-emulator/blob/master/examples/SingleSwitch/SingleSwitch.ino#L49) , where the user would need to supply a method for:

1. Checking if the source is still available
2. Reading the next byte from the source

The rest of the inputs I think could be simple parameters.

rusboard, 2017-01-10:
Will you merge it with master? Or will continue work in this branch?

witnessmenow, 2017-01-10:
For the moment I will continue on this branch. Feel free to raise a PR on that branch if you are looking to make changes (assuming you can do that)

rusboard, 2017-01-10:
> I tried with a more normal image there and it was incredibly slow

Did you try disable Serial prints when sending image? Serial print may slow down this operation.
Maybe it help a little to speed up upload.

I can't test sendImage, because I don't have SD module for arduino. I ordered one today :)
I tried save image from web to Wemos, but no luck and in internet no info, how to save image from web to arduino.

closed 2017-01-29 by witnessmenow