witnessmenow/Universal-Arduino-Telegram-Bot #11: "last_message_recived" property should be "last_message_received"

Nzbuu, 2017-01-04:
There's a spelling mistake in the UniversalTelegramBot API. I can fix it but that will affect break existing programmes that use it. What do you prefer?

witnessmenow, 2017-01-04:
Darn my spelling strikes again! Thanks for pointing it out.

I think it would be good to fix, it will fail to compile for people who update and still have old code but hopefully its easy enough to find.I guess the earlier it's fixed the less people it will impact. 

If you would like to submit a pull request making this change (including updating the examples) that would be awesome! I can pick it up at some stage if you can't though

witnessmenow, 2017-01-06:
Closed by #12 pull request which has been merged

closed 2017-01-06 by witnessmenow