witnessmenow/Universal-Arduino-Telegram-Bot #19: BulkMessages example is not compiling on Travis with Platformio (Compiling fine in Arduino IDE)

witnessmenow, 2017-01-09:
As per title

[Here is an example of a failed build](https://travis-ci.org/witnessmenow/Universal-Arduino-Telegram-Bot/jobs/190111729)

> /tmp/tmpfRZHsn/src/BulkMessages.ino: In function 'ArduinoJson::JsonObject& getSubscribedUsers(ArduinoJson::JsonBuffer&)':
> /tmp/tmpfRZHsn/src/BulkMessages.ino:113:36: error: 'class ArduinoJson::JsonBuffer' has no member named 'parseObject'
> JsonObject& users = jsonBuffer.parseObject(file_content);

I'm guessing the platformio config or Travis config needs to be tweaked

rusboard, 2017-01-09:
I have same error, when updated yesterday ArduinoJson library to 3.8.0, I rolled back to 3.7.3 again.
Will try fix later for 3.8.0. I don't know what changed in library.

witnessmenow, 2017-01-09:
The platformio is using the latest arduinojson library 5.8.0 but an earlier one can be specified if needed

rusboard, 2017-01-09:
Oh, my mistake about v3. Yes, 5.8.0 not working with my example. But 5.7.3 working.
I will later find out, where is problem and what need to fix.

witnessmenow, 2017-01-09:
Thanks! Let me know if you need anything

witnessmenow, 2017-01-09:
Fixed by @rusboard in #21

closed 2017-01-09 by witnessmenow