witnessmenow/Universal-Arduino-Telegram-Bot #21: Create a public Telegram group for users of the library

witnessmenow, 2017-01-09:
It might be helpful to help each other via Telegram, we are obviously all users of it!

rusboard, 2017-01-09:
Try to create it. We will try to join there :)

rusboard, 2017-01-10:
@universal_arduino_telegram_bot
https://telegram.me/universal_arduino_telegram_bot

Something like this?

rusboard, 2017-01-10:
Universal Arduino Telegram Bot library logo (ver 0.01) created too for channel :)

rusboard, 2017-01-10:
Group is removed. You can create your own group with same username and url and you will be as creator there.

rusboard, 2017-01-10:
I already have couple question. How to fork send_image branch, can't understand.

witnessmenow, 2017-01-10:
Created now, I used a shorter name to make it easier to share

http://t.me/arduino_library

witnessmenow, 2017-01-29:
New URL https://t.me/arduino_telegram_library

closed 2017-01-29 by witnessmenow