witnessmenow/Universal-Arduino-Telegram-Bot #34: Support the additional inline Keyboard types

witnessmenow, 2017-01-29:
Currently inline keyboard only supports URLs, additional work is needed to handle callbacks

prassaschr, 2017-04-18:
Kindly requesting the "callback button" type.
Thank you!

witnessmenow, 2017-04-19:
I won't get to make any changes to the library for a few weeks at least. I don't think there would be a lot of changes needed, the getUpdates message will need to support the callback.

I created a branch to do some work on the getUpdates to support channel posts, I guess that would be a good starting point.

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/Channel_Post/src/UniversalTelegramBot.cpp#L338

I think if that is modified to check for CallbackQuery you should be able to parse whatever is needed from there

`else if (result.containsKey("callback_query")) { `

[It should contain the following fields](https://core.telegram.org/bots/api#callbackquery)

If you wanted to try it out that would be great!

witnessmenow, 2017-04-19:
See #37 for more info on that branch

witnessmenow, 2017-10-16:
Available in the V1.1.0 RC branch

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/V1.1.0

closed 2020-09-30 by witnessmenow