witnessmenow/Universal-Arduino-Telegram-Bot #37: Support for Channel_post

littleyoda, 2017-02-11:
Support for Channel_post would be great.

see https://core.telegram.org/bots/api-changelog#november-21-2016

An Example of the result
{
   "ok":true,
   "result":[
      {
         "update_id":895636948,
         "channel_post":{
            "message_id":161,
            "chat":{
               "id":-10012345678,
               "title":"Title Example",
               "type":"channel"
            },
            "date":1486841742,
            "text":"Example Text"
         }
      }
   ]
}

littleyoda, 2017-02-11:
And a second example ("/bottest test test")
```
{
   "ok":true,
   "result":[
      {
         "update_id":895636959,
         "channel_post":{
            "message_id":275,
            "chat":{
               "id":-1001090227340,
               "title":"Iot",
               "type":"channel"
            },
            "date":1486843652,
            "text":"/bottest test test",
            "entities":[
               {
                  "type":"bot_command",
                  "offset":0,
                  "length":8
               }
            ]
         }
      }
   ]
}
```

mblwb, 2017-02-12:
just use @channel_name as chat_id.

littleyoda, 2017-02-12:
This works only for sending emails to channel that are public. 

But in this case you still don't get any messages back, because getUpdates searches for messages only in the section results/messages and not in the section results/channel_post.

witnessmenow, 2017-02-12:
Hey @littleyoda thanks for raising the issue. 

Adding the example response here is very helpful so thanks for that too!

Yeah this is something we could handle for sure, at the moment bot only cares about messages but this needs to be extended handle other types from getUpdates such as this one.

I've never seen the entities object before, I'll have to take a bigger look into it

If created a new branch over the next few days supporting this could download it and try it out?

littleyoda, 2017-02-12:
If you want to reproduce this kind of messages:
- Create a channel
- Add the Bot on this channel
- Make the Bot Administrator of this channel
- Change the privacy modus of this bot to disable via /setprivacy
- Send a Message

Of course, I will be happy to test every new source code.

witnessmenow, 2017-02-12:
I did some work towards this there. It requires a good bit of code clean up but is working for me

[It's on this branch](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/Channel_Post)

I added a "type" that you can check to see is it a channel post so you can handle it differently if required. I also added "chat_title" if that is useful. 

Example usage below

`
        if(bot.messages[i].type == "channel_post"){
          bot.sendMessage(bot.messages[i].chat_id, bot.messages[i].chat_title + " " + bot.messages[i].text, "");
        } else {
          bot.sendMessage(bot.messages[i].chat_id, bot.messages[i].text, ""); 
        }
`

littleyoda, 2017-02-12:
Just tested and it works fine.

Perfect. Thanks

witnessmenow, 2017-10-16:
Available in the V1.1.0 RC branch

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/V1.1.0

closed 2020-09-30 by witnessmenow