witnessmenow/Universal-Arduino-Telegram-Bot #38: Too long messages put the library into an endless polling loop

NDscham, 2017-02-13:
**Issue:** 

Sending messages longer than `maxMessageLength `defined in [UniversalTelegramBot.h](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/UniversalTelegramBot.h) sends the device into an endless polling loop where it polls the same message Id till infinity.

**Proposed Fix**:
Cut off message and terminate at `maxMessageLength` in  `UniversalTelegramBot.cpp::sendGetToTelegram`:

```c++
if (mess.length() >= maxMessageLength)
{
	mess = mess.substring(0,maxMessageLength-5);
	mess = mess + "\"\}\}\]\}";
}
```

For full code see attached modified cpp.
[UniversalTelegramBot.cpp.txt](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/files/771441/UniversalTelegramBot.cpp.txt)

witnessmenow, 2017-02-13:
Thanks for raising the issue.

Interesting solution to the problem, but you could not guarantee you would be at that level in the jsonArray. 

Maybe counting the { versus the } would be one way to figure it out.

I think the max message length can be bumped much higher than it is. I made a call to the reddit api before on the esp8266 and it could handle a much longer response than that.

NDscham, 2017-02-13:
Increasing the JsonBuffer length increases processing time, which might be undesirable. I absolutely agree forcibly terminating a message is not the best solution (and the way I did it is unflexible), but it's probably better than an infite loop or skipping this message (by incrementing message_id).
Dynamically counting/closing the Array is probably better. Note, that there are {} **and** [].

witnessmenow, 2017-02-13:
Increasing the message length shouldn't slow down processing time (well
other than when the message is longer than your old max). Its just a random
enough variable inherited from on of the other libraries this was based on.

Yeah for sure we need to do something about it, entering loops is the worst
case scenario. I think maybe a combination of both might work well.

I would like to make message length configurable, so adding the Json closer
first might be a good idea

I'll take a look this week if I get time.

NDscham, 2017-02-13:
Awesome, thanks! I'll could look into it as well, although I am not really familiar with JSON.

karonth, 2017-02-28:
One possible solution to these problems is to return -1 as the message number in case the json decoding fails. If you do a getUpdates?offset=-1 it'll flush the queue skipping the message. 
The problem with raising the buffer size is that'll be always a message that's bigger...so it won't be a complete solution. In many applications loosing a message is better than an infinite loop.

denzen84, 2017-10-21:
Probably I found solution. 
Function `int UniversalTelegramBot::getUpdates(long offset)` in some cases returns undefined result due to bug. 

As you see, the last **return** might never called.

`   if (response != "") {
    // ..................... code here
    return 0;
  }
}`

efeuentertainment, 2017-11-08:
i've experienced a similar loop.
i'm unsure if this is the same issue as @NDscham described.

when my esp8266 tries to send a long message, he sends the same message about 5 times.
after sendMessage(), the printed "response" was cut. only the last ~100-300 characters of the response were received/printed. and thus `checkForOkResponse()` failed to find `{"ok":true`
this resulted in a loop inside `sendPostMessage()` which sends the same message about 5 times.
basically this code inside `sendPostMessage()`:
```
while (millis() < sttime+8000) { // loop for a while to send the message
      [...]
      sent = checkForOkResponse(response);
      if (sent) {
        break;
      }
    }
```
i don't know why "response" gets cut.

and a possible fix could be to automatically split long messages into multiple smaller ones, but that's not the best solution i believe.

springm, 2018-04-03:
I would love to see an automatic cut at the defined maximum message length very much. 

I use the library for simple automation tasks, but recently got spammed by - I guess - other bots and due to my lack of skill to skip those messages, my program didn't work any more until I increased the maximum message size and read all those unwanted messages to clear the message queue

springm, 2018-04-03:
And thinking it further, a flushMessageQueue action would be useful - I'd call it after boot to make sure my Arduino can start with a clean slate. Unfortunately I am no C-Programmer, otherwise I'd try to implement it myself

witnessmenow, 2018-04-04:
@springm you can do this with bot.getUpdates if you just pass in a large
number, it will basically mark all messages that telegram currently knows
about with a smaller message id as processed. At least I think that should
work

ema800, 2021-08-24:
@springm To automatic prevent infinite loop,  in `UniversalTelegramBot.cpp` if you modify `UniversalTelegramBot::getUpdates` adding:

```
if (response.length() >= maxMessageLength) { 
    last_message_received = -2;
    closeClient();
    return 0;
}
```

In the next `bot.getUpdates()` since are you going to add +1 to `bot.last_message_received` variable that is set to -2, it will become equal to -1. When the update_id is -1 Telegram server returns the update_id of the last message received. So if there is another message after the long one, then it will read it instead of get stuck in infinite loop.