witnessmenow/Universal-Arduino-Telegram-Bot #39: secure one to one telegram communication

mkeyno, 2017-02-15:
hi , sorry to open ticket for my question but I really appreciated to have your advise. Actually I intend to make secure one to one communication between my ESP module and my phone (or couple of allowed phones), There is ESP8266WebServer, hosting web page, and intend to register allowed phones numbers by it, what is your suggestion to find id number of allowed phones and made secure communication
Thanks

witnessmenow, 2017-02-15:
Hey, thanks for your message, no worries about opening an issue!

The best way of doing this is to check either the chat_id or the from_id of the message you are dealing with and compare against known values.

from_id is always the telegram of the user that sent the message, even if its in a group chat.

chat_id is the chat where the message came from, so if I'm talking to the bot directly it will be my id , but if I send a message from a group it will be the groups id .

What I had for one project and I thought it worked well is a group chat that was allowed control the esp8266 and I just checked was the bot.messages[i].chat_id == knownValue

With this only people you messaged from that group were allowed send commands, and if I wanted to add more people I just added them to the group

Does this help?

mkeyno, 2017-02-15:
thanks @witnessmenow   for your advice but honestly  after opening this ticked I've noticed Telegram bot available in public so anyone can send command to my module and filtering authorized message probably chock my little ESP , after some sneaking around I found it is the best to webhook the bot to third party server which  can filter and authenticated the user message and to get new update it is just need websocket client service connected to this remote server

witnessmenow, 2017-02-15:
Yeah it is technically open for anyone to send messages to, but this would
control who it replies to.

Bots can't talk to other bots so it would have to be people manually
sending your bot messages, which I'm not sure why anyone would do that,
especially if its not replying.

Sounds like you got sorted though, which is great!

rusboard, 2017-02-16:
Do authorization - it's 15-20 minute work.
You can look "BulkMessages" example and see how work with file system. And keep in file admins user ids. And before accept command check user_id from Telegram with admin in file (or in buffer, like variable).

mkeyno, 2017-02-16:
thanks @rusboard  but my module should do much important task than filter the bot message or band unauthorized user  ,I did some test with fast sending message and notice true user message easily lost , I think it is good way to put user authentication and message parsing on third party server and put  the new update on client websocket line, its hard way because I should deal with php code and mysql handling which is not my expertise

mkeyno, 2017-02-16:
@witnessmenow  your example work fine , but when I try to use it in my sketch crash due to certificate's validity, I've tried to synced time but, it wont help , do you have idea what might be issue? the discussion about this strange issue continue  in [ gitter room ](https://gitter.im/esp8266/Arduino), why don't  you join

mkeyno, 2017-02-17:
@witnessmenow   I open ticket for that matter , would you please take look
https://github.com/esp8266/Arduino/issues/2975

Bingo600, 2017-03-19:
I just did basic auth with from_id
I'm allowing 2 phones (chat id's) to be authenticated
I left the debug prints in there , as this is a demo.

`
#define CHAT_ID_1   "xxxxxxxxx"
#define CHAT_ID_2   "yyyyyyyyy"

String allowed_id[]={       CHAT_ID_1 ,
                            CHAT_ID_2,
                            "" // Terminate witl Empty string (Null length)
                         };

// --- Telgram functions ---

void handleNewMessages(int numNewMessages) {
  for (int i=0; i<numNewMessages; i++) {
    String chat_id = String(bot.messages[i].chat_id);
    String from_id = String(bot.messages[i].from_id);
    String text = bot.messages[i].text;

    String from_name = bot.messages[i].from_name;
    if (from_name == "") from_name = "Guest";

    //char msg[255];
    //sprintf(msg,"Cname=%s - ID=%s - Text= %s ",from_name,chat_id,text);
    //Serial.println("Cname=%s - ID=%s - Text= %s ",(const char*)from_name,(const char*)chat_id,(const char*)text);
    //Serial.println("Cname=%s - ID=%s - Text= %s ",String(from_name),chat_id,text);
    Serial.println("Cname="+from_name+" - ChatID="+chat_id+" - FromID="+from_id+" - Text="+text);

    int allowed = 0;
    
    for(int j=0; allowed_id[j].length() != 0; j++)
    {
      if(allowed_id[j] == from_id)
        allowed = 1;
    }
    
    if(allowed == 1)
      Serial.println("Allowed="+from_id);
    else
      Serial.println("NotAllowed="+from_id);

    if(allowed != 0)
    {
      if (text == "/red") {
        turnLEDRed();
        bot.sendMessage(chat_id, "LED is now RED");
      }
  
      if (text == "/blue") {
        turnLEDBlue();
        bot.sendMessage(chat_id, "LED is now BLUE");
      }
  
      if (text == "/green") {
        turnLEDGreen();
        bot.sendMessage(chat_id, "LED is now GREEN");
      }
  
      if (text == "/toggle") {
        toggleOnBoardLED();
        bot.sendMessage(chat_id, "Toggled onboard LED, status is now " + String(onboardLedState));
      }
  
      if (text == "/ldr") {
        int value = checkLDR();
        bot.sendMessage(chat_id, "LDR reading is " + String(value));
      }
  
      if (text == "/options") {
        String keyboardJson = "[[\"/red\", \"/green\", \"/blue\"],[\"/toggle\"],[\"/ldr\"]]";
        bot.sendMessageWithReplyKeyboard(chat_id, "Choose from one of the following options", "", keyboardJson, true);
      }
  
      if (text == "/start" || text == "/help") {
        String welcome = "Welcome to Universal Arduino Telegram Bot library, " + from_name + ".\n";
        welcome += "The following options are available.\n\n";
        welcome += "/red : to switch RBG LED red\n";
        welcome += "/green : to switch RBG LED green\n";
        welcome += "/blue : to switch RBG LED blue\n";
        welcome += "/toggle : to toggle onboard LED\n";
        welcome += "/ldr : to return the current value of the LDR\n";
        welcome += "/options : returns a custom reply keyboard\n";
        welcome += "/help : displays this message again\n";
        bot.sendMessage(chat_id, welcome, "Markdown");
      }
    } //end if(allowed ...
  } //end for(...
}
`

witnessmenow, 2017-05-11:
I dont think there is anything we can do on the library side for this. I'll raise a new issue for providing an example like @Bingo600 example

closed 2017-05-11 by witnessmenow