witnessmenow/Universal-Arduino-Telegram-Bot #41: Support for Yùn board

Wentstorm, 2017-06-12:
I was trying to make it work with my Yùn board, I figured it was just a matter of using BridgeSSLClient instead of WiFiSSLClient, but I didn't succeed to make It work.
Enabling debug I get:
```
GET Update Messages
.... connected to server
got response
handleNewMessages
2027 
```
(the last value might vary)
At this point I have no idea on how to make it work.

witnessmenow, 2017-06-12:
I think to debug we'll have to add a few more serial prints through out the library, but I suspect the issue is with the json library. Esp8266 works well with a dynamic buffer, maybe the yun should use a static one. 

Could you add some more prints to the library to check where exactly it's failing? 

I'm hoping to get a Arduino 101 soon that I would be able to test with, hopefully that would be closer to the yun. 

I would like to be able to help you more with supporting the yun but it would be too expensive for me to buy one to try get it working myself (it's over £50 for me to get one,  even they are $30 from AliExpress ) 

I would be happy to try help you in any way I can though! 

I would focus on the getUpdates method, but serial prints before and after each stage in there an we would have a better idea what is wrong then

Wentstorm, 2017-06-12:
Uploading different code (EchoBot) and uncommenting
Serial.write(c); into the library I get several rows in loop:
`GET Update Messages
.... connected to server
<html>
<head><title>400 The plain HTTP request was sent to HTTPS port</title></head>
<body bgcolor="white">
<center><h1>400 Bad Request</h1></center>
<center>The plain HTTP request was sent to HTTPS port</center>
<hr><center>nginx/1.10.1</center>
</body>
</html>

port</center>
<hr><center>nginx/1.10.1</center>
</body>
</html>
`
I'm tryng to find what causes: HTTP request was sent to HTTPS port

witnessmenow, 2017-06-12:
Hmm, 

Could you try this library out and see if it works? The way I do the requests on it are slightly different and I want to see does it have an impact

https://github.com/witnessmenow/arduino-coinmarketcap-api

You don't need any api keys or anything for it so it should be quick to port

witnessmenow, 2017-06-12:
Just reading your edit now, cool, that must mean that it reads the message OK. 

Keep me posted!

Wentstorm, 2017-06-12:
I will try the other Api as soon.
Now I figured out that sendGetToTelegram(command) in getUpdates returns an empty string, so I looked into sendGetToTelegram and realized that you don't get into while (client->available()) ...
I basically have no idea on what's happening so far :(

witnessmenow, 2017-06-12:
It could be to do with the way the request is done, would be interesting to
see if the other api works ok

Wentstorm, 2017-06-12:
The other Api  doesnt'work, basically is tryng to parse an empty string

witnessmenow, 2017-06-12:
Do you have any examples of the bridge ssl client working? Would be
interested to see what is different

Wentstorm, 2017-06-12:
I found I can get some values from coinmarketcap using plain Http instead of ssl (and BridgeClient instead of SSL client), still get error in parsing Json but I get legit values:

```
        "id": "dogecoin", 
        "name": "Dogecoin", 
        "symbol": "DOGE", 
        "rank": "18", 
        "price_usd": "0.00334156", 
        "price_btc": "0.00000120", 
        "24h_volume_usd": "20739400.0", 
        "market_cap_usd": "366809204.0", 
        "available_supply": "109771844357", 
        "total_supply": "109771844357", 
        "percent_change_1h": "0.42", 
        "percent_change_24h": "-6.12", 
        "percent_change_7d": "-4.42", 
        "last_updated": "1497276246", 
        "price_eur": "0.0029791344", 
        "24h_volume_eur": "18490004.676", 
        "market_cap_eur": "327025078.0"`

```

Now I will try the same trick with telegram

witnessmenow, 2017-06-12:
Telegram only supports https so I don't think it will work

Wentstorm, 2017-06-12:
At this point I'm almost sure that the problem is in the SSLClient, when I use it I don't get any response from the server, when I don't use at least I get An error page. Guess what's wrong with this

jotadepicas, 2017-11-13:
Hi everyone, any updates on this? 
I'm looking into making a telegram bot using Dragino Yun Shield + Arduino UNO.
Thanks!

witnessmenow, 2018-01-26:
I don't have a YUN and probably won't be getting one so I am unable to provide any support with this, if someone else has any luck please report back here!