witnessmenow/Universal-Arduino-Telegram-Bot #42: Add support to send text files.

nymJ, 2017-06-13:
It would be great to send text files. For example: temperature log files.

witnessmenow, 2018-01-26:
If someone is interested in implement this I would be happy to accept in a PR.

It should re-use the sendMultiPartForm data that is already implemented so it should not be too difficult to add (hopefully!)

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/UniversalTelegramBot.cpp#L143

Telegram bot API details here:
https://core.telegram.org/bots/api#senddocument

unocerobits, 2018-03-30:
I have tried to sending files as OTA in #78, this is a dirty code, but would be great pull this function (files or OTA would be great)
But I don't know how do I handle to make the OTA update, but sending and receiving is possible.

witnessmenow, 2019-06-07:
From Malik on the Group chat:

I just followed what you refer to 
added this on the cpp file :  

String UniversalTelegramBot::sendDocumentByBinary(
    String chat_id, String contentType, int fileSize,
    MoreDataAvailable moreDataAvailableCallback,
    GetNextByte getNextByteCallback) {

  if (_debug)
    Serial.println("SEND Document");

  String response = sendMultipartFormDataToTelegram(
      "sendDocument", "document", "file.txt", contentType, chat_id, fileSize,
      moreDataAvailableCallback, getNextByteCallback);

  if (_debug)
    Serial.println(response);

  return response;
}

then followed sendphotoSD example 
another thing I did was setting the "content_type" which is in my case "document/txt",so calling the method in Arduino IDE was like this : 
String sent = bot.sendDocumentByBinary(chatId, "document/txt", myFile.size(), isMoreDataAvailable,getNextByte);

Josh-Reimer, 2024-06-14:
> It would be great to send text files. For example: temperature log files.

I would really appreciate a feature like this as well. Can someone explain how to achieve the same with the existing library? (without modifying the libraries code just in the Arduino code)

Josh-Reimer, 2024-06-14:
Also, does the send photo from sd card work with the littlefs filesystem?