witnessmenow/Universal-Arduino-Telegram-Bot #43: i have a problem :(

nimaaryamehr, 2017-06-13:
what is this ERRor?

In file included from FlashLED.ino:3:0:
J:\Program Files\Arduino\libraries\Universal-Arduino-Telegram-Bot-master\src/UniversalTelegramBot.h:27:25: fatal error: ArduinoJson.h: No such file or directory
 #include <ArduinoJson.h>
                         ^
compilation terminated.
Error compiling.

witnessmenow, 2017-06-13:
Install Arduino json from the library manager

Sorry for short response!

nimaaryamehr, 2017-06-13:
so tnx
i install that but yet have this problem :/

Veneth, 2017-10-12:
After I install and include library i get this error
Arduino: 1.6.5 (Windows 8.1), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/common/include/nm_common.h:45:0,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/socket/include/socket.h:60,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/socket/include/socket_buffer.h:45,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/WiFiClient.h:26,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/WiFiSSLClient.h:23,
                 from MyCustomKeyboard.ino:10:
C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/bsp/include/nm_bsp.h:110:23: error: conflicting declaration 'typedef long unsigned int uint32'
 typedef unsigned long uint32;
                       ^
In file included from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include/ets_sys.h:11:0,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/pgmspace.h:12,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.h:29,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:26,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Stream.h:26,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../StringTraits/ArduinoStream.hpp:12,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../StringTraits/StringTraits.hpp:31,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../JsonVariantComparisons.hpp:10,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:11,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../JsonVariant.hpp:16,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:15,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/JsonParser.hpp:10,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/JsonBufferBase.hpp:10,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson.hpp:10,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson.h:10,
                 from MyCustomKeyboard.ino:1:
C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include/c_types.h:29:29: error: 'uint32' has a previous declaration as 'typedef unsigned int uint32'
 typedef unsigned int        uint32;
                             ^
In file included from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/common/include/nm_common.h:45:0,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/socket/include/socket.h:60,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/socket/include/socket_buffer.h:45,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/WiFiClient.h:26,
                 from C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/WiFiSSLClient.h:23,
                 from MyCustomKeyboard.ino:10:
C:\Users\Elcot\Documents\Arduino\libraries\WiFi101-master\src/bsp/include/nm_bsp.h:133:22: error: conflicting declaration 'typedef long int sint32'
 typedef signed long  sint32;
                      ^
In file included from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include/ets_sys.h:11:0,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/pgmspace.h:12,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.h:29,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:26,
                 from C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Stream.h:26,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../StringTraits/ArduinoStream.hpp:12,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../StringTraits/StringTraits.hpp:31,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../JsonVariantComparisons.hpp:10,
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:11,
[... cut ...]
                 from C:\Users\Elcot\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson.h:10,
                 from MyCustomKeyboard.ino:1:
C:\Users\Elcot\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include/c_types.h:32:29: error: 'sint32' has a previous declaration as 'typedef int sint32'
 typedef signed int          sint32;
                             ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

witnessmenow, 2017-10-16:
@nimaaryamehr is this still an issue?

@Veneth please open a new issue if you are still having problems, but it looks like you are using the WiFi101 examples on an ESP8266, use the examples under the ESP8266 folder

witnessmenow, 2018-01-26:
No responses, closing

closed 2018-01-26 by witnessmenow