witnessmenow/Universal-Arduino-Telegram-Bot #44: How do I set the password?

nimaaryamehr, 2017-07-28:
I want to set up a password to control the robot
That means asking the password when start the robot
and then control led??!

witnessmenow, 2017-07-28:
You are better off to handle authentication using telegram ids

If you message @chatinfo_bot on telegram you can get your chat id or a
groups chat id. This is a unique Id that no one else will be able to have.

You can check in the handle new messages method if the command was from a
recognised person

It should be something like this

https://github.com/witnessmenow/Immersion-Bot/blob/master/Immersion-Bot.ino#L128

nimaaryamehr, 2017-07-28:
It's a good way
but I do not want to reprogram esp any time i change my telegram

rusboard, 2017-07-28:
Make auth file.
Send command to bot, /auth password (where password some dkjfgi2435u4)
If your ESP see, that this password is correct, then write this CHAT ID in file of allowed users

And when someone sending command to bot, you will need check his chat ID with chad ID in file.
And that's all.

rusboard, 2017-07-28:
As example you may use this demo
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/BulkMessages/BulkMessages.ino

There is already code how create file and add user there.
You just need check password before add user in file.

iranweld, 2017-08-29:
My Dear
How is possible i use bottoken saved on eeprom from address 0 to 50
I have problem for #Define Bottoken

witnessmenow, 2017-10-16:
Original Question answered I think 

@iranweld please open a new issue if you still need support

closed 2017-10-16 by witnessmenow