witnessmenow/Universal-Arduino-Telegram-Bot #45: Response message

SandTB, 2017-07-28:
Hi, i've the following problem, i need that the bot when receive a command wait for another message, and process it. How i can do that, Thanks in Advance.

rusboard, 2017-07-28:
Elementary, My Dear Watson.
You need make a flag for this (variable).

For example:
bool i_am_waiting_second_command = false;

When you receive first commander and you know that after this command you need wait second command, then make this flag TRUE;

And after next received message from TELEGRAM you need chack, if this flag TRUE, then it's second command for previous command. If this flag false, it's some main command.

After you will recognize, that second command for previous command was received make flag FALSE again.

I hope it will helps.

SandTB, 2017-07-28:
Thanks, then there isn't an "in-library" method, i should use the old method of the variable?

witnessmenow, 2017-10-16:
No, there is so little memory to play around with that custom features like this are left up to the consumer of the library to implement as not everybody will need them!

Thanks

closed 2017-10-16 by witnessmenow