witnessmenow/Universal-Arduino-Telegram-Bot #46: Issues Sending Messages

ianhill1, 2017-08-07:
Hello,

I am trying to get EchoBot to work, but bot.sendMessage() isn't sending messages. 

I checked the boolean return of sendMessage; returns true. EchoBot is otherwise working -- I am able to print the bot's last received messages to the serial monitor. 

Any help would be much appreciated! 
Thanks,
-Ian

rusboard, 2017-08-07:
Please attach your sketch file.

ianhill1, 2017-08-07:
Thanks for the fast reply! I am using nodemcu esp8266.
[EchoBot_notsending.zip](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/files/1206058/EchoBot_notsending.zip)

rusboard, 2017-08-07:
Are you sure that you using valid bot token?
In code I don't see nothing badly.

Go to your Telergam library directory and change in UniversalTelegramBot.h file bool _debug = false to true
And look into serial monitor. Maybe you will see something.

ianhill1, 2017-08-07:
The bot token should be valid, because I am able to access and print the bot's last received messages to the serial monitor. 
I ran the debugger, sent a few messages to my bot, and the following was printed to my serial monitor:

_incoming message length330
Creating DynamicJsonBuffer

got response
SEND Post Message

responding with:  Maybe?
to chat id:  424286303
GET Update Messages
.... connected to server

{"ok":true,"result":[]}

incoming message length23
Creating DynamicJsonBuffer

no new messages

Thanks again.

witnessmenow, 2017-10-16:
@ianhill1 is this still an issue for you?

witnessmenow, 2018-01-26:
No response

closed 2018-01-26 by witnessmenow