witnessmenow/Universal-Arduino-Telegram-Bot #53: Use advantage of Telegram's POLL mode in func "getUpdate"  to provide more efficiency

denzen84, 2017-10-12:
Hi,

I use this bot trough mobile 3G connection and I found that it eats a lot of mobile traffic.
I investigated this misadvantage and found the solution.

Telegram provides POLL mode of the connection. We should use special option in the query to say how long we can wait for the data.  In current release bot send update query every 1 second and immediately gets back empty message. To solve it, I modified the code of bot:

1. In header "**UniversalTelegramBot.h**" add  new definition "_#define POLL_TIME_WAIT 60 // seconds_"
Higher value is better for traffic economy

2.  In source "**UniversalTelegramBot.cpp**" add some modifications:

- function **UniversalTelegramBot::sendGetToTelegram(String command)**:
main loop:
while (millis() - now < **(POLL_TIME_WAIT*1000+1000)**) {....}

- function **UniversalTelegramBot::getUpdates(long offset)**
String command = "bot"+_token+"/getUpdates?offset="+String(offset)+"&limit="+String(HANDLE_MESSAGES)**+"&timeout="+string(POOL_TIME_WAIT)**;

I successfully tested these code improvements.

Thanks for your attention.

P.S. I'm too lazy to make new fork.

denzen84, 2017-10-12:
[src.zip](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/files/1378709/src.zip)

witnessmenow, 2017-10-16:
Available in the V1.1.0 RC branch

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/V1.1.0

denzen84, 2017-10-17:
IMHO default value for POLL should be 5-20 seconds.  
IMHO ~1Gb of spurious traffic every month for each ESP where this bot installed is very expensive for the World. Think about green trees and nature pollution.

witnessmenow, 2017-10-17:
It changes the default behavior of the library if any poll is included,
anyone who upgrades from v1 to v1.1 shouldn't have behaviours changed if it
can be helped.