witnessmenow/Universal-Arduino-Telegram-Bot #56: ESP8266 crashes when used with low RAM/Heaps available. EXCEPTION (29)

efeuentertainment, 2017-10-20:
Hello
I ran into a problem which i tried to investigate in the last week.
I'm using an ESP8266, wemos D1mini, and was able to reproduce the crash using the PhotoFromURL example.
This submitted issue might probably be a dublicate of #51 and #52 . Feel free to move and modify my post in any way you see fit.
Also note that this issue might be caused by ArduinoJSON or another library which Universal-Arduino-atelegram-Bot relies upon.

The crash occurs when sending messages. In my project the ESP crashes after sending 2-5 messages. Using the unmodified PhotoFromURL example it takes about 100-300 messages within ~3 minutes, which resulted in an Exception (29) crash.
I slightly modified the example and filled the RAM with a big global char array and wrote ESP.GetFreeHeap() to the serial output, which triggers the crash after sending ~10 messages.

The exception always occurs when the free heap size decreses beneath ~6500-5700.
The free heap size increases automatically when ~3min have passed. I assume the old buffers or something gets deleted when they are older than ~3min.
I enabled the _debug bool in UniversalTelegramBot.h

This is the modified sketch:
[telegram_exceptionist.zip](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/files/1401880/telegram_exceptionist.zip)

### Here 3 pssible combinations and their outcome:
**telegram_exeptionist.ino using WiFiClientSecure without filling RAM:**
_Sketch uses 296285 bytes (28%) of program storage space. Maximum is 1044464 bytes.
Global variables use 40484 bytes (49%) of dynamic memory, leaving 41436 bytes for local variables. Maximum is 81920 bytes.
Uploading 300432 bytes_

free heaps:
at start 17300 B
**crashes after 6160 B**
(this was almost impossible since 100-300 messages have to be sent, but the RAM starts to free up again after ~3min.)

<details><summary>Serial output: [expand]</summary><a>
GET Update Messages
.... connected to server

{"ok":true,"result":[{"update_id":72978882,
"message":{"message_id":1980,"from":{"id":xxxxxxxx,"is_bot":false,"first_name":"Dejan","last_name":"Lauber","language_code":"en-US"},"chat":{"id":xxxxxxxx,"first_name":"Dejan","last_name":"Lauber","type":"private"},"date":1508349619,"text":"/status","entities":[{"offset":0,"length":7,"type":"bot_command"}]}}]}

incoming message length357
Creating DynamicJsonBuffer

got response
handleNewMessages
1
free heap: 6344 B

msg length: 298
sendMessage start: 8789721 ms
SEND Post Message

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff0ee0 end: 3fff1500 offset: 01a0

>>>stack>>>
3fff1080:  3fff03f4 000013d6 000013d6 40223beb  
3fff1090:  00000208 3fff560c 3fffad2c 40223d84  
3fff10a0:  00000104 00009ca0 00001394 3fff560c  
3fff10b0:  00000040 00000082 3fffad14 40224036  
3fff10c0:  00800000 00000080 00000000 00000041  
3fff10d0:  00000040 0000003f 40205f50 00000041  
3fff10e0:  3fffb6bc 3fff560c 3fff6844 3fff48b4  
3fff10f0:  3fffbbd0 ffffffff 3fff560c 3fff560c  
3fff1100:  00000040 3fff560c 3fffad14 3fff560c  
3fff1110:  00000040 3fff560c 3fffad14 40224b92  
3fff1120:  3fff579c 0000003f fdd3d074 00000001  
3fff1130:  3fff560c 3fffa51c 3fffad14 00000001  
3fff1140:  00000010 3fff560c 0000000f 40224ded  
3fff1150:  3fff6844 3fff489c 0000000f 00000001  
3fff1160:  00000001 3fffa6c4 3fff28c9 40223c50  
3fff1170:  00000100 3fffa1a4 3fff282f 00000000  
3fff1180:  00000100 3fffa1a4 3fff282f 40225e79  
[... 347 lines cut ...]
0x40205fc0: operator delete(void*) at /home/brian/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/abi.cpp line 57
0x40100718: cont_norm at /home/brian/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S line 109

</a></details>

### Build options:
Board: "Wemos D1 R2 & mini"
CPU Frequency: "80MHz"
Flash Size: "4M (3M SPIFFS)"
Upload Speed: "921600"

### Other things i've tried:
Different USB Power adapter. no effect.
I got the same crash on another ESP8266, with different power adapter and cable. 
470uF Capacitor is in place on the 5V rail.
Added a 2200uF Capacitor to the 3.3V rail. no effect.
changed to 1M flash size. no effect.
changed clock to 160mhz. no effect.

If anyone has a clue on how to resolve this issue, or an idea on what to try next, let me know.

Best regards,
dejan lauber

PS: The ESP8266 and this telegram library blew my mind! Awesome work!

efeuentertainment, 2017-10-20:
i was thinking if that space is reserved for the wifi stack and causes the ecxeption.
i found this issue: [How to increase heap size by reducing dynamic memory size?](https://github.com/esp8266/Arduino/issues/3031)
where jjssoftware writes:
>where stack grows/shrinks at one end of this free RAM and heap grows/shrinks at the other end. If things get so bad where stack and heap meet/overlap due to RAM depletion, you get an exception and a reboot.

i wonder if it's possible to check the wifi stack size...?

apart from that, i checked the addresses "3FFF3***" from the serial monitor and they seem to be well within range of the user RAM "dram1". can't be an out-of-range issue then.
>3FFE8000h-3FFFC000h 	dram0 	14000h 	RAM 	RW 	User data RAM. Available to applications.
source: [Memory Map](https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map)

efeuentertainment, 2017-10-26:
**UPDATE:**
Stack size is by default 4KB. its defined in <cont.h>.
[this link includes an example on getting free stack](https://github.com/esp8266/Arduino/issues/2557)

when i run that everything looks normal. stack size: 4KB and free stack around 1000B-1400B.

but if the free heap size gets lower than 4KB it would cause a crash. (i assume)
**i printed the free heaps on various places within UniversalTelegramBot.cpp:**

a successfull sendMessage free heap size goes:
```
free heap: 6240 B
SEND Post Message
free heap: 5008 B
free heap: 4888 B
sending header  
free heap: 4704 B  
```
which is quite close to 4KB..
and a crash goes:
```
free heap: 6056 B
SEND Post Message
free heap: 4824 B
free heap: 4704 B
sending header
```
and then it crashes in UniversalTelegramBot.cpp in function "UniversalTelegramBot::sendPostToTelegram" on exactly this line (est. 74):
```
	// Connect with api.telegram.org
	if (client->connect(HOST, SSL_PORT)) {
```
which means its quite certainly not a bug in UniversalTelegramBot, but rather somewhere else.

[i did find another thread with a very similar issue in the esp8266 core](https://github.com/esp8266/Arduino/issues/2967)
where OrakelKSL writes:
>Every time client receives a request, the heap goes down and does NOT recover!
>...ir recovers, but it takes MINUTES!
>... and if you keep f5 pressed exception 28 or 29 occours.

and devyte's response:
>The fact heap would recover hints at sockets being kept in TIME_WAIT state, probably for a the typical 2 minutes. That is per RFC (I forget which), although for tiny embedded systems implementations usually don't hold to it.
Still, with latest git I don't currently see the issue when refreshing a webpage.

if it's the same issue i see little i can do about it...
i can either try to find where those sockets get put in TIME_WAIT, and change it to delete unused sockets,
or i do a dirty fix where i only update the UniversalTelegramBot if there is more than ... eg. 8KB free Heaps..

someone another idea?

gon0, 2017-11-03:
If it helps, I also got Exception(29) with the telegram bot, shown here: https://github.com/Aypac/Arduino-TR-064-SOAP-Library/issues/2#issuecomment-341767372

witnessmenow, 2017-11-03:
@gon0 making a https connection uses 18k of heap memory, I presume you are out of memory by the time out get down to the telegram line

gon0, 2017-11-03:
@witnessmenow Yeah, seems like that. 
However, you did very good work with the library! In my code, the Telegram-Bot is the main function, so I will leave out the https://github.com/Aypac/Arduino-TR-064-SOAP-Library library until there is a solution to use both.

efeuentertainment, 2017-11-04:
**UPDATE**
i found a workaround to clean up sockets kept in TIME_WAIT.
issue: [memory leak 184 bytes at a time #1923](https://github.com/esp8266/Arduino/issues/1923)
where psy0rz commented:
>since that pull request isnt merged yet and it will  take some time to reach upstream, here's a workaround that actually  cleanups the TIME_WAITS and frees memory:
```
// do this before including other stuff like wificlient.h
#include "lwip/tcp_impl.h"

void tcpCleanup()
{
  while(tcp_tw_pcbs!=NULL)
  {
    tcp_abort(tcp_tw_pcbs);
  }
}
```
>update: This goes in your main file. Call tcpCleanup regulary, for example from your loop() function.

it sounds as if there's a pull request with a better solution which will fix this issue in the future.
anyway, my sketch works for now!

in my case the ESP crashed when free heap went under ~4.5KB. but that's not always the same it seems.
[as OrakelKSL wrote:](https://github.com/esp8266/Arduino/issues/2967#issuecomment-341884180)
>Heap goes rapidly down to a few bytes [...]

which implies his heap went down to near zero...

closed 2017-11-04 by efeuentertainment

morrowwm, 2018-02-06:
FWIW, I updated my Arduino IDE to use the latest ESP8266 library following instructions here: [https://github.com/esp8266/Arduino#using-git-version](https://github.com/esp8266/Arduino#using-git-version)

Now my sloppy mashup of the excellent Universal-Arduino-Telegram-Bot and a webserver based on this other excellent code: [https://tttapa.github.io/ESP8266/Chap16%20-%20Data%20Logging.html](https://tttapa.github.io/ESP8266/Chap16%20-%20Data%20Logging.html)
is running fine now. 

Previously, it crashed repeatedly on `client->connect()` calls.