witnessmenow/Universal-Arduino-Telegram-Bot #57: SIM800 support

EgorChadov, 2017-10-20:
Hi! It is possible add support of SIM800 module in this library? I think it will be useful feature, because it can use gprs connection.

witnessmenow, 2017-10-21:
I think it might support it using the tinygsm library

https://github.com/vshymanskyy/TinyGSM/blob/master/README.md

The telegram library uses the standard Arduino client interface for
communicating with the network, and the tinygsm library provides a client
wrapper around these modules

I think if you instialise a TinyGsmClientSecure object as shown in the
https example of gsmtiny and pass it in the constructor of the telegram bot
as the client, it should in theory work

I don't have a sim800 to test so I haven't been able to try it out. If you
do try please let me know if it works!

EgorChadov, 2017-10-21:
Thank you for reply!
Yes, it works! Tried to send a simple message using UniversalTelegramBot + Tinygsm libraries and message come to my chat. Very excited with that.

I think you can write that your library works with sim800l+tinygsm. Thanks for help!

witnessmenow, 2017-10-21:
Awesome! Would you mind creating a pr with an example, or even if you could
share it here I could add it.

Thanks!

Where did you get your module?

EgorChadov, 2017-10-22:
Sure, I can create example, but only simple. I'm still very new with it.

But I have another problem now. I try your ReplyKeyboard Example with Tinygsm and that two libraries take much space. Arduino IDE says that "Not enough memory, the program may not work reliably". And then really program work not properly.

EgorChadov, 2017-10-22:
`#define TINY_GSM_MODEM_SIM800
#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>
#include <UniversalTelegramBot.h>

#include <SoftwareSerial.h>
SoftwareSerial SerialAT(8, 9); // RX, TX

TinyGsm modem(SerialAT);

const char apn[]  = "";   //apn
const char user[] = "";   //username
const char pass[] = "";   //pass

// Initialize Telegram BOT
#define BOTtoken ""  // your Bot Token (Get from Botfather)

TinyGsmClientSecure client(modem);
UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

void setup() {
  Serial.begin(9600);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(3000);

  modem.sendAT(GF("+CIPSSL=1"));    //enable ssl. maybe don't need
  modem.waitResponse();

  if (!modem.hasSSL()) {
    Serial.println("SSL is not supported by this modem");
    //delay(10000);
    return;
  }

  //wait for network
  Serial.print(F("Waiting for network..."));
  if (!modem.waitForNetwork()) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  //connect to gprs
  Serial.print(F("Connecting to "));
  Serial.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println(" fail");
    delay(10000);
    return;
  } else {Serial.println(" OK");}

  //not needed actually
  IPAddress local = modem.localIP();
  modem.waitResponse();
  Serial.println(local);
}

void loop() {
  String chat_id = ""; 
  String text = "hello!";
  String parses = "";

  Serial.println("Send message");
  Serial.println(bot.sendMessage(chat, text, parses));

  delay(50000);
}

`

Very simple code I used for test how it works. In future, I can add other examples if it needs.

I got this module from aliexpress. Need link? Also, you can buy it on eBay.

witnessmenow, 2017-10-22:
What board are you using? Something like an uno or a nano might not have enough ram for Arduino json

EgorChadov, 2017-10-22:
Ok. I got it.
I'm using uno and nano for my project.

witnessmenow, 2017-10-22:
Something with more ram might work better, like even an esp8266 even if you
don't use WiFi

EgorChadov, 2017-10-22:
Ok. Thank you for helping me! 👍

witnessmenow, 2017-10-22:
No problem @EgorChadov , I was wondering if it would work before since I seen the TinyGSM library, so really happy that you confirmed it did.

Would you mind linking to the one you got? The individual module is pretty cheap, but it looks like it would be hard to work with and then all the boards seem kind of expensive!

EgorChadov, 2017-10-23:
I have two versions. 1 - http://ali.pub/1y0ha9, 2 - http://ali.pub/1y0hex
But you must pay attention to this module. It needs 3.3V for supply and may consume about 2A at peak. Otherwise, it may have a problem with work.

I hope it information comes handy for you!

witnessmenow, 2017-10-23:
Thanks Egor!

Do you know anything about the sim900? I think the 800 only supports
2g,while the 900 supports 3g as well, but I'm not sure if it works the same
as the 800!

My cell provider is 3g/4g only. You can get 2g providers in Ireland but it
would be easier for me if it was 3g

EgorChadov, 2017-10-23:
I think it the same modem that sim800 but it has 3g support.

I have chosen 800 because it cheaper and for arduino project doesn't need speed connection. 2g make it good as well.  
But in your case, I think you need 900. Look this option - http://ali.pub/1y0wjy. It more expensive but it sim900 :)

enunugraha1, 2017-11-28:
Is it possible to use SIM 900 and use another library (not tinyGSM library)?, that I understand is the point is to make connections to internet and so I can just set SIM 900 to do as gprs modem use existing library
Thanks.

witnessmenow, 2017-11-28:
Why don't you want to use tinygsm?

The telegram library requires a httpclient that supports https, in theory
any client that matches this should work

enunugraha1, 2017-11-28:
Try once with this sketch, the SIM 900 didn't turn on automatic must be push SIM 900 power button to turn on, and there is error massage like "GSM modem not selected" tinyGSM should work with SIM 900, shouldn't it?

EgorChadov, 2017-11-28:
Are you sure that SIM900 support https? As far as I know only sim800 and sim808 support ssl. 

You can buy sim800l. It's not very expensive. At [this ](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/57#issuecomment-338761693) post I gave two options.

enunugraha1, 2017-11-29:
I think SIM900 need firmware update to make it support HTTPS, but after searching for no one have success to do that, HTTPS AT command still get error massage. Your suggestion is better choice I can implement SIM800L + tinyGSM lib, if anyone could add more examples for that it would be great

rahmathasan, 2018-01-07:
Hi, @EgorChadov , i have try to verify the code (above) with arduino ide, but there is error compiling. 
"Error compiling for board Arduino/Genuino Uno"

i want to build a telegram bot with arduino using sim900 or sim800L gsm module.
thank you

witnessmenow, 2018-01-26:
If someone can submit a PR for this against the V.1.1 branch providing an example for this I will happily include it in the next version (which I will release in a couple of weeks)

Timmae3000, 2018-03-04:
Hi anyone.

I want to build an telegram bot with a arduino and a sim800 gsm module.
i've testet something with the sketch from  EgorChadov from this thread.

the first message by the bot was send successfully.  Great :)

But not all messages goes out or come in to the bot.
So in the terminal i can see, that all 5 seconds a message was send.
But in the chat with the bot it comes only 1 message per minute in.
Telegram Bot can send 30 messages per minute.

When i start the arduino and a message was send i can restart the arduino and an new message was send...so i can do it 6times per minute and become 6 messages in the chat...

So anyone has an idea? :)

jimmisitepu88, 2018-04-12:
Dear @EgorChadov ,
I trying to compile your program:
`#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>
#include <UniversalTelegramBot.h>

#include <SoftwareSerial.h>

SoftwareSerial SerialAT(8, 9); // RX, TX

TinyGsm modem(SerialAT);

const char apn[]  = "";   //apn
const char user[] = "";   //username
const char pass[] = "";   //pass

// Initialize Telegram BOT
#define BOTtoken ""  // your Bot Token (Get from Botfather)

TinyGsmClientSecure client(modem);
UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

void setup() {
  Serial.begin(9600);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(3000);

  modem.sendAT(GF("+CIPSSL=1"));    //enable ssl. maybe don't need
  modem.waitResponse();

  if (!modem.hasSSL()) {
    Serial.println("SSL is not supported by this modem");
    //delay(10000);
    return;
  }

  //wait for network
  Serial.print(F("Waiting for network..."));
  if (!modem.waitForNetwork()) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  //connect to gprs
  Serial.print(F("Connecting to "));
  Serial.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println(" fail");
    delay(10000);
    return;
  } else {Serial.println(" OK");}

  //not needed actually
  IPAddress local = modem.localIP();
  modem.waitResponse();
  Serial.println(local);
}

void loop() {
  String chat_id = ""; 
  String text = "hello!";
  String parses = "";

  Serial.println("Send message");
  Serial.println(bot.sendMessage(chat, text, parses));

  delay(50000);
}`

but we get problem when compile,

In file included from C:\Users\JimmiSitepu\Documents\OneDrive\KERJAAN\MIKROAVR.COM\Running Project\Polmed\Jodi_team\sim800_telegram\sim800_telegram.ino:1:0:

C:\Users\JimmiSitepu\Documents\Arduino\libraries\TinyGSM-master\src/TinyGsmClient.h:64:4: error: #error "Please define GSM modem model"

   #error "Please define GSM modem model"

    ^

exit status 1
Error compiling for board Arduino Nano.

what I have to fix? Thanks
Thanks

EgorChadov, 2018-04-20:
Hello @jimmisitepu88, sorry for long answering.

Try add modem model after including TinyGsm library.

`// Select your modem:
#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590`

Add #define and your modem. But as far I know it works only with 800 and 808 for now.

jimmisitepu88, 2018-06-07:
Thanks a huge @EgorChadov , its work...

u-l-m-i, 2018-10-21:
Hello @EgorChadov,

may I ask if you also managed to receive messages with your telegram bot & SIM800, because I'm only able to send messages, but not receive messages.

So if possible it would be really great if can provide an example 
>
> Very simple code I used for test how it works. In future, I can add other examples if it needs.
> 

Best regards

ipungdc, 2019-03-17:
Dear @EgorChadov ,

I tried your program and defined sim800L but there was an error. what's the solution ?
thanks...

jimmisitepu88, 2019-03-17:
I get error to when compile that's code,.when downgrade Arduino json to.5, compile not error, but telegram bot not working

Dapatkan Outlook untuk Android<https://aka.ms/ghei36>

ipungdc, 2019-03-18:
I've downgraded it, but the code still error

gunaone, 2020-04-03:
any one was sucses with this sim800..?
plase share..

Belma2010, 2020-07-05:
> ```
> #include <TinyGsmClient.h>
> #include <ArduinoHttpClient.h>
> #include <UniversalTelegramBot.h>
>
> #include <SoftwareSerial.h>
> SoftwareSerial SerialAT(8, 9); // RX, TX
>
> TinyGsm modem(SerialAT);
>
> const [...]

I already use your code and add botToken and chat_id but i still didn't get the messages from the bot. Serial print "send messages" and "0", but connect to gprs is work. Did i miss any step?

witnessmenow, 2020-07-05:
As far as I know the sim800 does not support TLS V1.2, so will not work with telegram anymore

closed 2020-07-05 by witnessmenow

fbi1671, 2021-10-10:
try the SSLClient by https://github.com/govorox

I got the esp32 to send message the telegram group, but still figuring how to retrieve message from the group

asus1305, 2022-01-15:
@fbi1671 i have error for this #include "SSLClient.h", #include "ca_cert.h". can u help me about this library?

fbi1671, 2022-01-15:
hi - please send me your code, let me try to check it.

haxord7hasib, 2022-03-27:
any hope for A9G gsm module ?

CIumbia44, 2022-12-04:
Has anyone made it work with SIM7600?

mr-bobur, 2023-07-06:
> Has anyone made it work with SIM7600?

Hi, I am also struggling to use lilygo_sim7600G_esp32 board and telegram bot, you find something useful?