witnessmenow/Universal-Arduino-Telegram-Bot #61: How to tell if bot is valid

AcuarioCat, 2017-12-28:
Hi,
is there a way to know that the bot is valid when it is created?
I would like to know that the token is valid and that I have a valid bot that I can use and if not then prompt for a valid token.
Thanks.

witnessmenow, 2017-12-28:
Hi,

The easiest way is probably to use bot.getMe(); this will return true if
everything worked OK and will return false if it didn't (although it could
have failed for other reasons than your not token, internet connection for
example)

AcuarioCat, 2018-01-15:
Thanks, as you say not perfect but it is the best solution available. Thanks.

closed 2018-01-15 by AcuarioCat

burdeyniyua, 2019-04-04:
I use next solution:
```
if (text == "ping) {
bot.sendMessage(chat_id, "I am Alive!", "Markdown");                          
  }
```