witnessmenow/Universal-Arduino-Telegram-Bot #64: wifiManager.setSTAStaticIPConfig

plcabgut, 2018-01-24:
hello
when i use wifimanager with static ip 

IPAddress ip          (192, 168,   1, 59);
IPAddress gateway(192, 168,   1,  1);
IPAddress mask     (255, 255, 255,  0);

wifiManager.setSTAStaticIPConfig(ip, gateway, mask);

the esp8266nodemcu dont receives nothing
if I let wifimanager in dhcp mode dhcp then it works fine and receives every command fro  telelegram
```

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

//WIFIMANAGER
#include <DNSServer.h>            //Local DNS Server used for redirecting all rs to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

#define BOT_TOKEN_LENGTH 46
char BOTtoken[BOT_TOKEN_LENGTH] =  "----MYTOKEN----";  

WiFiClientSecure client_secure;
UniversalTelegramBot *bot;

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

IPAddress ip     (192, 168,   1, 59);
IPAddress gateway(192, 168,   1,  1);
IPAddress mask(255, 255, 255,  0);

void handleNewMessages(int numNewMessages) {
  for (int i = 0; i < numNewMessages; i++) {
    String chat_id = String(bot->messages[i].chat_id);
    String text_bot=bot->messages[i].text.substring(1,bot->messages[i].text.length());
    if (text_bot.equals("X")) {
      bot->sendMessage(chat_id, "X", "");
    }
    if (text_bot.equals("Y")) {
      bot->sendMessage(chat_id, "Y", "");
    }

    if (text_bot.equals("start") || text_bot.equals("help")) {
      String help= "/X \n";
      help+= "/Y";
      bot->sendMessage(chat_id, help, "");
    }
  }
}

void setup() {
  Serial.begin(115200);
  WiFiManager wifiManager;
```
**

> //IF I COMENT NEXT LINE IT WORKS FINE OTHERWISE DONT RECEIVES NOTHING

**
```
  wifiManager.setSTAStaticIPConfig(ip, gateway, mask);
  wifiManager.autoConnect("AP_door", "pwd");
  if (WiFi.status() == WL_CONNECTED) {
    bot = new UniversalTelegramBot(BOTtoken, client_secure);
    Serial.println("WiFi connected");
    Serial.print(WiFi.localIP());
    Serial.println("/");
  } 
}

void loop() {
  if (millis() > Bot_lasttime + Bot_mtbs)  {
    int numNewMessages = bot->getUpdates(bot->last_message_received + 1);
    while(numNewMessages) {
      Serial.println("got response");
      handleNewMessages(numNewMessages);
      numNewMessages = bot->getUpdates(bot->last_message_received + 1);
    }
    Bot_lasttime = millis();
  }
}
```

witnessmenow, 2018-01-24:
Is your gateway definitely set correctly?

This would be more of an issue for the WiFi manager library than this one

plcabgut, 2018-01-24:
ok i'll post this question there if i get a solution i'll tell you

witnessmenow, 2018-01-26:
Not a question for this library, going to close.

Good luck!

closed 2018-01-26 by witnessmenow

evbelda, 2018-04-12:
I have the same problem with only ESP8266Wifi.h (not wifiManager) but staticIP

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

IPAddress ip (192, 168, 1, 59);
IPAddress gateway(192, 168, 1, 1);
IPAddress mask (255, 255, 255, 0);

void setup()
{
WiFi.mode(WIFI_STA);
WiFi.config(ip, gateway, mask);  // (DNS not required)
WiFi.begin(ssid, password);

....

}.....

Same code for Telegrambot ubove....

int numNewMessages = bot->getUpdates(bot->last_message_received + 1); //numNewMessages is 0 everytime.

With DHCP works fine.

After last universalTelegrmbotUpdate 1.1.0

Please, clould you check this?

Hardware: Webmos D1 R2

witnessmenow, 2018-04-12:
Can you provide a full sketch that I could test? You don't need to provide
a bot token.

igorelterrible, 2019-01-31:
sorry to ask here...

but i´ve got a simular situation & can´t find much info  out there -> exist a solution to use WiFi.config with this lib (without the wifi manager)?

thank´s

plcabgut, 2019-01-31:
tomorrow i will ask