witnessmenow/Universal-Arduino-Telegram-Bot #65: bot.getUpdates -- many, many questions

jandrioli, 2018-01-25:
First of all thanks for sharing your library. I am using it on a ESP32 dev module (ESP-WROOM-32). 
I'm having a lot of issues with this method bot.getUpdates() because it generates a 

> core 1 panic'ed: LoadProhibited

with a bunch of register dump. 
This is very easy to reproduce. I am using your example code with my own bot token. 
I simply leave the chip running in a loop, polling for new updates every 5 seconds. Eventually, like after about 3 or 4 minutes, I will receive updates for an inexisting telegram message. 

In order for me to explain this better, I looked into the code of bot.getUpdates() and I noticed a couple of "interesting" things:
1 - the update code *always* brings 1 update at a time, no more. So why are we using an array?
2 - this function bot.getUpdates returns the amount of new updates found (if there are any). And this is *always* going to be 1
3 - the method posting to telegram.org is using the 

> define HANDLE_MESSAGES 1

to limit the amount of updates that we receive everytime. But if ever something goes wrong with this request, or we get more than one update, the function getUpdate will crash. 

This being said, let's look at the sequence of events causing my crash. 
1 - loop forever, call bot.getUpdates() every 5 seconds
2 - return value of bot.getUpdates() is always zero
3 - after about 3 or 4 minutes, return value of bot.getUpdates is 1073492764 !!! Bingo

As you can see that return value should never exceed 1 but in this case it does. My assumption is that something in the Bot Heap is getting screwed up. 
I am using the library available from the Arduino libraries manager. 
Do you have any tips as to where I should start debugging first ?
What could be wrong ?
I have tried this code on a 8266 ESP-12F and have not had any issues whasoever.

witnessmenow, 2018-01-25:
Hey!

The original plan was to allow support of multiple messages, but it doesn't
seem like boards have enough memory for it.

Sorry I don't have a lot time to reply right now, but take a look at the
PRs of this library, someone has pushed changes to make it work better with
the ESP 32, I haven't had a chance to test yet though.

I would be interested to hear how you get on with it!

witnessmenow, 2018-01-25:
Also join us in the group chat if you'd like!

https://t.me/arduino_telegram_library

jandrioli, 2018-01-25:
Hi and thanks for this quickest response.
I had a look into that telegram group but I have to say, it's very difficult to follow what's going on there. 
Not a good channel for developpers in my opinion; so I will continue to post here on github. 

I enabled debugging and checked the PR from PRO2XY, his code did solve problems with SSL. In fact the only thing to add is the call to close the connections. 

I am currently testing this on multiple devices. So far it's working very well.

AlfredoJLM, 2018-10-09:
Hello jandrioli.

I have the same problem that happed to You.

While the code works perfectly with the module esp8266- with esp32 module bot.getUpdates () it normally returns 1073492764.

I really appreciate if you can guide me on how to call to close the connections in arduino.

ManieCR, 2021-03-25:
> First of all thanks for sharing your library. I am using it on a ESP32 dev module (ESP-WROOM-32).
> I'm having a lot of issues with this method bot.getUpdates() because it generates a
>
>
> with a bunch [...]

Good day, I know its been a while since you've created this post, but we're you able to solve the problem with bot.getupdates returning a large and then crashing your code? I suspect I have the same problem with my two ESP8266-07's also using telegram... You mentioned that the get.updates problem didn't occur on your ESP8266-12F, well I do happen to have two new Esp8266-12F's still unopened, should I replace my 07's with the 12F's??? Or is other solution to get around the get.updates problem???
Thanks Manie

jandrioli, 2021-04-02:
Hi, the root cause of this problem was related to SSL. As you know Telegram
uses secured api which depends on encryption.
All arduino and similar boards unfortunately had this problem. Not only
ESP. You have that HEAP issue because the http requests are failing.
I cannot remember exactly, IIRC it was just a question of changing some
variables in my code. Perhaps change the telegram bot constructor to use a
secure http or something like that.
Check out that line *botclient.setFingerprint*
Last time I touched this was 2018 so take your time to investigate. Here is
a little bit of code example from my sketch:
```
#include <ArduinoJson.h>
#include "FS.h"
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266SSDP.h>
#include <UniversalTelegramBot.h>
#include <TimeLib.h>
#include "esp.h"
ESP8266WebServer      server ( 80 );
WiFiClientSecure      botclient;
UniversalTelegramBot  bot(BOTtoken, botclient);

. . . some bla bla bla and then later in setup() i have these lines . . .

  wifi_set_event_handler_cb(wifi_event_handler_cb);
  setup_wifi();
  yield();
  botclient.setFingerprint("my_personal_HEXADECIMALS_here"); <-- google it
  bot.sendMessage(g_tgCHAT, "Bot is running.", "HTML");
```