witnessmenow/Universal-Arduino-Telegram-Bot #66: Support ESP32

witnessmenow, 2018-01-26:
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/pull/62

Need to fix tests on it and Add some examples, but looks like it is working well for people

Darkhub, 2018-01-29:
Hi, I'd like to help with ESP32 version, the current one isn't working correctly and after some time throws an exception and crashes:
[...]
got response
got response
got response
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
[...]

witnessmenow, 2018-01-29:
Hi @Darkhub , did you try the one linked to #62 ?

Its not currently in the main library yet, you could try download it from here and see does it work for you

https://github.com/PRO2XY/Universal-Arduino-Telegram-Bot/tree/esp32_port

Darkhub, 2018-01-29:
I was using your library, my bad. :/
I am now using the PRO2XY library and the provided example for ESP32 seems working good, I'll do some modifications and leave it running for some days to test long-term stability.

As a note, I was experiencing a strange thing, ESP32 was not able to connect to wifi network, every other example worked but not the one using Telegram library.
It seems that commenting out this line solved the problem and ESP32 connected on first try:
//WiFi.mode(WIFI_STA);

JacoFourie, 2018-02-14:
Hi. 

I am testing the ESP32 version now. I get multiple messages at random. What could it be ?

Darkhub, 2018-02-14:
@JacoFourie multiple messages? From?

JacoFourie, 2018-02-15:
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/pull/62

JacoFourie, 2018-02-23:
I found it. https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/74