witnessmenow/Universal-Arduino-Telegram-Bot #67: i have a w5200 Ethernet shield

mnickhoo, 2018-01-30:
hi I have a W5200 Ethernet shield and I haven't wifi module and I want to use my shield instead wifi and my question is how to use W5200 + UniversalArduinoTelegramBot ?

witnessmenow, 2018-01-30:
It probably doesn't work, I haven't seen an Ethernet shield that supports
HTTPS which is required by telegram

mnickhoo, 2018-02-05:
telegram use https only for webhook method but I want use a getUpdates method that won't need https

witnessmenow, 2018-02-05:
All requests must be HTTPS:

https://core.telegram.org/bots/api#making-requests

"All queries to the Telegram Bot API must be served over HTTPS"

mnickhoo, 2018-02-05:
you right but you can make a request to telegram bot API with a getUpdates method only on HTTP protocol

witnessmenow, 2018-02-05:
Weird! You are right it does work!

[image]

Interesting, I think I'll add a functionality to change the Port number that it connects to (so it can be changed to 80 manually). Doing this along with passing in a non HTTPS client should make it work with without using the HTTPS, will have to test to confirm though.

I still think you will have troubles running it on an Uno though, I think it won't have enough memory to parse the response into JSON

mnickhoo, 2018-02-05:
yeah I have Ethernet shield and want to use your library to connect my bot on telegram messenger

evbelda, 2018-03-26:
Please. Can you put the part of the code where you can change the port to 80 manually? is in a part of the library or in the main code? is it  in the json library?

witnessmenow, 2018-03-26:
I tested it and it didn't work for me.

But if you want to do your own testing: change the port number in the
header file of the library.

Then in your sketch pass in a http client into the constructor of the bot
(for Ethernet shield the client seems to be called EthernetClient)

I don't think an uno will have the ram to process responses even if the
http call works