witnessmenow/Universal-Arduino-Telegram-Bot #68: bot.sendPhotoByBinary Not Working

HoseinPorazar, 2018-02-03:
Serial.println("was successfully sent");  is called but nothing is sent to telegram.

gilang4official, 2018-05-08:
I have same problem, 
Serial.println("was successfully sent"); is called

but the photo didn't sent to my telegram.

tuan-karma, 2018-11-27:
I have the same problem on ESP32 (SPIFFS).

mitofile, 2019-04-21:
I have the same problem but reading file from spiffs. Is this issue related to (https://github.com/esp8266/Arduino/issues/2281)

WolfEnders, 2020-01-02:
**ESP32 TTGO T-Camera-Plus (board 'ESP32 Dev Module' with PSRAM)**
To me the issue happened sporadically when using SPIFFS, while it happened permanently when sending an image directly from the camera buffer "camera_fb_t *fb". 
So perhaps a timing thing at Telegram?

In UniversalTelegramBot.cpp UniversalTelegramBot::sendMultipartFormDataToTelegram(..) I added after each "client->print" / "client->println" / "client->write" a "delay(50)".

Now it works reliable, even with shorter delay @10ms. But @1ms again it fails big time. 
Therefore out of paranoia, I kept 50ms.

robotzero1, 2020-02-04:
@WolfEnders 
> **ESP32 TTGO T-Camera-Plus (board 'ESP32 Dev Module' with PSRAM)**

How did you get this working? I've been going round in circles trying to chunk the data. I hacked about with sendMultipart to send a QVGA file in one go but anything bigger doesn't arrive.

witnessmenow, 2020-02-04:
I'm playing around with this at the moment, I will be updated soon

WolfEnders, 2020-02-05:
> 
> 
> @WolfEnders
> 
> > **ESP32 TTGO T-Camera-Plus (board 'ESP32 Dev Module' with PSRAM)**
> 
> How did you get this working? I've been going round in circles trying to chunk the data. I hacked about with sendMultipart to send a QVGA file in one go but anything bigger doesn't arrive.

As I briefly mentioned as an **ugly workaround**:
I literally added after every "client->write" in method sendMultipartFormDataToTelegram a "delay(5)".

I really appreciate the effort, that @witnessmenow is dealing with this subject.

javierferwolf, 2020-03-07:
@WolfEnders hi, I have the same problem! you said that in your case the problem was solved add after every "client->write" in method sendMultipartFormDataToTelegram a "delay(5)". but i use the v1.2.0 and i added that you indicated but I still have the same problem!  will there be any additional arrangements?
I use 8266 and I want to send the photo from SPIFFS!

javierferwolf, 2020-03-16:
@WolfEnders @witnessmenow hi, I have the same problem! you said that in your case the problem was solved add after every "client->write" in method sendMultipartFormDataToTelegram a "delay(5)". but i use the v1.2.0 and i added that you indicated but I still have the same problem! will there be any additional arrangements?
I use 8266 and I want to send the photo from SPIFFS!

WolfEnders, 2020-03-16:
Perhaps ESP32 makes the difference.

jameszah, 2020-05-20:
Hi, trying to implement this with mixed success on an esp32 with quite a bit of other code running at different priorities.  It worked the first time (for 10 pictures), then started failing regularly as I fiddled with it for a couple hours - reading from sd_mmc files files, and stdio files, and reading from the fb buffer.  The following kludge got it working again -- based on the @WolfEnders idea, but without digging into the library.

```
bool isMoreDataAvailable() {
  return (fb_length - currentByte);
}

uint8_t getNextByte() {
  currentByte++;
  if (currentByte % 100 == 0) delay(10);
  return (fb_buffer[currentByte - 1]);
}
```
It inserts a delay as the library is fetching bytes ???

Here are my current libraries
```
Using library ESP32 at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESP32 
Using library WiFi at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
Using library WiFiClientSecure at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFiClientSecure 
Using library UniversalTelegramBot at version 1.1.0 in folder: C:\Users\James\Documents\Arduino\libraries\UniversalTelegramBot 
Using library ArduinoJson at version 5.13.5 in folder: C:\Users\James\Documents\Arduino\libraries\ArduinoJson 
Using library ESPmDNS at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS 
Using library SD_MMC at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD_MMC 
Using library FS at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
Using library HTTPClient at version 1.2 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient 

```

jameszah, 2020-05-21:
I think the problem is in sendMultipartFormDataToTelegram.

After the sending is done it goes into this loop (below), where it gets some trash to set the responseReceived to true, but it has not got the response yet from telegram, and then it breaks out of the while, and closes the connection, before telegram has fully consumed all the data, and thus it abandons that picture.  It closes the connection an instant after the client->print(end_request);.

So the following debug statement after client->print(end_request); seemed to solve the problem

```
    client->print(end_request);
    if (_debug)
      Serial.print(end_request);

    Serial.print("... Done Sending.  Client.Available = ");  Serial.println(client->available());
    delay(2000);
    Serial.print("... 2 secs later.  Client.Available = ");  Serial.println(client->available());
    count = 0;`

```
Here is the while loop that closes the connection too quickly.
```
   while (millis() - now < waitForResponse) {
      while (client->available()) {
        char c = client->read();
        responseReceived = true;

... collect up the response

      if (responseReceived) {
        if (_debug) {
          Serial.println();
          Serial.println(body);
          Serial.println();
        }
        Serial.println("... breaking");
        break;
      }
    }
  }

  closeClient();
```

cvdabbeele, 2020-06-24:
Just FYI.  I experimented with  "client_tcp.writeln" abd "client_tcp.write" (as described here: https://www.viralsciencecreativity.com/post/esp32-cam-motion-alert-send-image-to-telegram) and I can confirm I see the same issues.  For a stable solution, I also need to add delays after a write....maybe this info can help you
chris

thejoshua3, 2020-06-27:
I encountered the same issue in my project. After trying version 1.2.0 of library (and the code in ESP32-Cam.ino) without resolving the problem, I turned back at current version of library (1.1.0) adding the delay after "client->print(end_request)". My two AI-Thinker ESP32-CAM after 4 days not missed the sending of a single photo not even. Great!

JulioCalandrin, 2023-10-28:
I faced the same issue on v1.3.0 and solved it by adding "client->flush();" after every client->write, client->print and client->println calls. Working smoothly :).