witnessmenow/Universal-Arduino-Telegram-Bot #72: Conflict: can't use getUpdates method while webhook is active

patrickelectric, 2018-02-09:
```bash
api.telegram.org
443
.... connected to server

{"ok":false,"error_code":409,"description":"Conflict: can't use getUpdates method while webhook is active"}

incoming message length107
Creating DynamicJsonBuffer

Response contained no 'result'
```

witnessmenow, 2018-02-09:
Hmm interesting. What would you expect to happen in this case?

patrickelectric, 2018-02-09:
I'm running all esp8266 examples and none is working.

witnessmenow, 2018-02-09:
Your must be used somewhere else that is using webhooks? Create a new bot or regenerate the token for it

witnessmenow, 2018-02-09:
Correction cause i cant edit on mobile: Your bot token must be used somewhere else that is using webhooks? Create a new bot or regenerate the token for it

patrickelectric, 2018-02-09:
Using a fresh one works.. ty !
Do you know what this ? My old bot is not running.

witnessmenow, 2018-02-09:
I've never used web hook but maybe it still active even when the bot is not
running. I think if you generate a new token for that bot it should work
again

closed 2018-02-09 by patrickelectric