witnessmenow/Universal-Arduino-Telegram-Bot #76: Delay time in bot.sendMessage (RFID interfaced with NodeMCU)

JulliardK, 2018-03-17:
I'm using RFID reader with NodeMCU. 
This is what it does: when MFRC522 reads a tag, it counts the total number of tags that have been read till now, then send the number via Telegram.

The issue i am facing is this:
The reader is able to read one tag after another without pause if I take out the following code:
`bot.sendMessage("242432855", hello, "");`

However with that code in, the reader is only able to read a new tag after 2-3 seconds.

Is there a way to circumvent this?

What i tried is moving this code around:
```
    mfrc522.PICC_HaltA();
    mfrc522.PCD_StopCrypto1();
```
The above code is supposed to prime the reader to read a new tag. But no matter where i place it (I tried putting it before the "bot.sendMessage" code), there will always be that 2-3 second delay as long as the bot.sendMessage is there.

This is the part of that code in question

```
  else {
    Serial.print(F("\n**Returned**\n"));
    totalNum = totalNum + 1;
    String hello = String(totalNum); 
    lockServo.write(180);
    delay(1000);
    lockServo.write(0);
    bot.sendMessage("242432855", hello, ""); //huge delay time

  }
    mfrc522.PICC_HaltA();
    mfrc522.PCD_StopCrypto1(); // for reader to be able to read new tags
}
```

JacoFourie, 2018-03-17:
In this case I would use a ESP32 that has 2 cores. Then use one core to count the tags and use the other core to send the messages and handle the wifi stuff. That is what I do. I let one core handle all the sensors and control an the other handles the UI and WiFi.

JulliardK, 2018-03-19:
Good advice. Thanks!

closed 2018-03-20 by JulliardK