witnessmenow/Universal-Arduino-Telegram-Bot #83: Any chance of send image from spiffs?

danbicks, 2018-06-23:
Hi would be brilliant to have an example that allows you to send an image stored in spiffs..

Is this possible?

Cheers

Dans

svenp, 2019-06-22:
Hi Dans, 
i had same Idea and here is the code for you....

```
readFile(SPIFFS, "/img.jpg");

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\r\n", path);

   file = fs.open(path);
    if(!file || file.isDirectory()){
        Serial.println("- failed to open file for reading");
        return;
    }
   
     sent = bot.sendPhotoByBinary(chat_id, "image/jpeg", file.size(),
            isMoreDataAvailable,
            getNextByte);
        if (sent) {
          Serial.println("was successfully sent");
        } else {
          Serial.println("was not sent");
        }
        
file.close();
}

```

witnessmenow, 2019-06-22:
If you are sending a static image that is not generated by the Arduino
itself, you might consider hosting it somewhere like Imgur and then use
sendPhotoByUrl

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/SendPhoto/PhotoFromURL/PhotoFromURL.ino

javierferwolf, 2019-12-25:
> Hi Dans,
> i had same Idea and here is the code for you....
>
> ```
> readFile(SPIFFS, "/img.jpg");
>
> void readFile(fs::FS &fs, const char * path){
>     Serial.printf("Reading file: %s\r\n", path);
>
>    file [...]

hi Sven,  I have a problem when I want to compile I get an error 
"exit status 1
'class String' has no member named 'size' "
What could the problem be?