witnessmenow/Universal-Arduino-Telegram-Bot #84: Heap Space

danbicks, 2018-06-26:
Has this build changed? when I run the basic example after a period of time there is a stack overflow and the ESP crashes. I have added to the example to return free heap, when I run this from telegram client on my phone I can see the heap keeps reducing until the ESP crashes.
The following hardware is a Nodemcu V3
Any fixes for this?

Cheers

Dans

`

// Code added to return heap free size from bot.

if (text == "/tech") {
        int StrHeap = (ESP.getFreeHeap());
        String msg = "Dans Node IOT Statistics.\n";
        msg += String(ssid) +" : " + String(WiFi.RSSI()) + " dBm\n";
        msg += "IP address : " + WiFi.localIP().toString() + "\n";
        //msg += "Vcc : " + String(ESP.getVcc()) + " mV\n";
        msg += "Bell Counter : " + String(kCounter) + "\n";
        msg += "uptime : " + String(millis()/(1000*60*60)) + " h\n\n";
        msg += "You are Registered!\n";
        msg += "This Chat ID is : " + chat_id + "\n";
        msg += "Free Heap Space : " + String(StrHeap) + "\n";
        bot.sendMessage(chat_id, msg, "");
         
      }

Debug console output when crash occurs.

Soft WDT reset

ctx: cont 
sp: 3fff1280 end: 3fff1700 offset: 01b0

>>>stack>>>
3fff1430:  3fff05ec 00000729 00000729 4010020c  
3fff1440:  3fff839c 00000045 3fff2cf4 00000000  
3fff1450:  3fff839c 00000045 3fff2cf4 401004d8  
3fff1460:  3fff3030 3fff3040 3fff3210 40206bbc  
3fff1470:  3fff839c 00000045 3fff2cf4 40203ee5  
3fff1480:  00000000 3fff6234 00000045 40217aa4  
3fff1490:  00030317 3fff6201 00000055 40217aa4  
3fff14a0:  00000001 00000017 00000040 40203f5e  
3fff14b0:  00000010 00000017 00000050 00000000  
3fff14c0:  3fff6234 3fff1510 00000001 40217b30  
3fff14d0:  00000001 3ffe8fdc 00000010 40217b30  
3fff14e0:  00000010 3fff03c0 3fff6234 3fff155c  
3fff14f0:  40243fe4 3fff03c0 3fff03c0 40203ae0  
3fff1500:  3fff15cc 3fffa458 00000003 40206ffc  
3fff1510:  3ffe9074 000004a5 3fff03c0 402062dd  
3fff1520:  3ffe8fdc 3fffa458 3fff03c0 402071d4  
3fff1530:  3fff15cc 3fffa458 3fff03c0 40206325  
3fff1540:  3fff15cc 3fffa458 3fff0324 40205442  
3fff1550:  00000000 00000000 00000000 000a3953  
3fff1560:  00000040 3fff15c0 3fff3fd4 0000000f  
3fff1570:  00000000 0000003c 3fff15c0 402066b7  
3fff1580:  3fff05ec 000013e6 3fff15c0 402066e9  
3fff1590:  3fff03c0 00000000 3fff15c0 402067f0  
3fff15a0:  3ffe91a4 00000000 3fff0324 000a3953  
3fff15b0:  3fffa458 00000000 3fff0324 40205640  
3fff15c0:  3fffbe44 0000003f 0000003c 3fff2ea4  
3fff15d0:  0000000f 00000000 3fff6144 0000003f  
3fff15e0:  0000003c 3fff165c 00000000 402045e0  
3fff15f0:  00000010 3fff1650 3fff1650 3fff165c  
3fff1600:  3fff0324 3fff1650 3fffa458 40205719  
3fff1610:  3ffe93d0 00001144 3fffa44c 00000200  
3fff1620:  3fff1668 3fff0320 3fff1650 40206736  
3fff1630:  3ffea0d4 3fff0320 3fff0330 3ffe8ae8  
3fff1640:  00000000 3fff0320 3fff0330 40202621  
3fff1650:  3fffad44 0000000f 00000000 3fffac6c  
3fff1660:  000000cf 000000c0 3fff6a7c 0000000f  
3fff1670:  00000009 3fff69a4 000000cf 000000c0  
3fff1680:  3fffb044 0000000f 00000004 3fffb02c  
3fff1690:  0000000f 00000005 3fff344c 0000000f  
3fff16a0:  00000009 3fff0324 3fff0610 402062dd  
3fff16b0:  00000001 3fff0314 3fff0610 40206308  
3fff16c0:  3fff2004 40202e64 3fff0610 3fff06d0  
3fff16d0:  3fff0320 3fff0324 00000001 40202919  
3fff16e0:  3fffdad0 00000000 3fff06c8 40206cf8  
3fff16f0:  feefeffe feefeffe 3fff06e0 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
`

MoonMoon82, 2019-07-22:
damn it .. I've got the same issue but one year later. And even on this post no one ever replied..
I was logging free heap bytes when I recognized it shrinks everytime I use bot.getUpdates .
Interestingly even on bot.sendMessage it shrinks for a short time and then after a minute it frees some bytes again.

witnessmenow, 2019-07-23:
Honestly, the library is inefficient from a heap perspective.

If I was doing the library again now I would not use String anywhere, but I
didn't really know what I was doing back then (and I still don't). I do not
have time for a major re write at the moment.

Something like this is implemented much more efficiently from a heap
perspective

https://github.com/witnessmenow/arduino-hackadayio-api

witnessmenow, 2019-07-23:
Also the above sample of code is also very inefficient from a heap
perspective, concatenating strings is the devil , reducing the amount of
that in the sketch will improve your heaps chances.

Reserving space up front might also help

https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/reserve/