witnessmenow/Universal-Arduino-Telegram-Bot #86: Is crashing all the time

BolverBlitz, 2018-06-30:
Exception (29):
epc1=0x4020ebaa epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff0440 end: 3fff0a00 offset: 01a0

>>>stack>>>
3fff05e0:  00820000 00000082 3fff5a8c 4021ba26  
3fff05f0:  3ffef984 3fff5864 3fff5a8c 4021bc20  
3fff0600:  3fff366c 00000208 3fff366c 40100670  
3fff0610:  00000104 00002738 000004e7 00000042  
3fff0620:  00000040 00000082 3fff1cb4 4021bf39  
3fff0630:  00800000 00000080 00000000 00000041  
3fff0640:  00020771 0000003f 3fff63dc 00000041  
3fff0650:  3fff3254 3fff1cb4 3fff63dc 3fff7ddc  
3fff0660:  3fff3768 ffffffff 3fff5864 4021c5b5  
3fff0670:  00000040 3fff5864 3fff1cb4 3fff5864  
3fff0680:  00000040 3fff5864 3fff1cb4 4021ca73  
3fff0690:  3fff591c 0000003f 47cd2c35 00000001  
3fff06a0:  3fff5864 3fff1c9c 3fff1cb4 00000001  
3fff06b0:  00000010 3fff5864 0000000f 4021ccc8  
3fff06c0:  3fff63dc 3fff607c 0000000f 00000001  
3fff06d0:  00000001 3fff2ae4 3fffa359 4021bacc  
3fff06e0:  00000100 3fff101c 3fffa357 00000000  
3fff06f0:  00000100 3fff101c 3fffa357 4021de51  
3fff0700:  3fff0730 00000000 000000d0 00000030  
3fff0710:  050fb848 3fff2af0 3fff63e8 00000004  
3fff0720:  3fffa351 3fff848c 3fffa351 40218b4c  
3fff0730:  ca4c0303 7af0c120 ccf4f799 e1ebb074  
3fff0740:  4f28a589 1acd51a8 57cd14db b913aff1  
3fff0750:  400e8641 a09094b0 fa719e4c b848d3aa  
3fff0760:  3fffa351 00000004 3fff848c 40218fe1  
3fff0770:  50b38fa5 f9833231 f4f44301 40216250  
3fff0780:  3885c32e 3fffc6fc 3ffef9e0 40211db5  
3fff0790:  00000000 00000004 00000004 402038f8  
3fff07a0:  00000013 000013d3 00000064 40207034  
3fff07b0:  0000004a 00000000 00000026 00000004  
3fff07c0:  3fffa351 00000004 3fff848c 00000004  
3fff07d0:  3fffa351 00000004 3fff848c 40218954  
3fff07e0:  39821fb0 000000d1 000000d1 00000027  
3fff07f0:  0000005c 3fffa351 3fff848c 40218d30  
3fff0800:  00000010 3fffa384 3fff848c 401004d8  
3fff0810:  3fff0868 0000000e 00000010 3fff82c4  
3fff0820:  4020374c 3fff848c 3fff0860 01000000  
3fff0830:  3fff0868 00016a49 3fff848c 40218abc  
3fff0840:  3fff848c 3fff15bc 3fff1974 3fff82c4  
3fff0850:  3fff3f6c 00016a49 3fff1974 40203f2d  
3fff0860:  3fff848c 3fff15bc 40206fbc 3ffef9e0  
3fff0870:  00003a98 00003a98 00003a98 4020136f  
3fff0880:  3ffef720 00000000 3fff3f6c 3fff0930  
3fff0890:  000001bb 3ffe8df8 3ffef720 4020417b  
3fff08a0:  000001bb 3ffef720 3ffe8df8 3fff0930  
3fff08b0:  000001bb 3ffef720 3ffe8df8 40204221  
3fff08c0:  401072a0 dca79a95 401072a0 dca79a95  
3fff08d0:  3ffe9a2c 3fff0940 3ffef684 40204431  
3fff08e0:  3fff8fbc 00000054 3fff0930 40206a7f  
3fff08f0:  3ffef984 000003b4 3fff0930 40206ab1  
3fff0900:  3fff094c 3fff094c 3fff0930 40206bb8  
3fff0910:  3fff094c 3fff094c 3ffef684 00000001  
3fff0920:  3fff094c 3fff094c 3ffef684 40206319  
3fff0930:  3fff2cd4 0000005f 00000054 4020755c  
3fff0940:  3fff8da4 0000000f 00000000 00000000  
3fff0950:  00000000 00000000 3fff8fbc 0000005f  
3fff0960:  00000054 00000000 00000000 00000000  
3fff0970:  00000000 00000000 00000000 40100690  
3fff0980:  3ffef6b4 3ffef684 3ffef9a4 00000001  
3fff0990:  3ffef714 3ffef684 3ffef9a4 402026a0  
3fff09a0:  00000000 00000000 00000000 00000000  
3fff09b0:  00000000 00000000 3ffef9a4 402025fc  
3fff09c0:  00000001 b200a8c0 feefeffe feefeffe  
3fff09d0:  feefeffe feefeffe feefeffe 3ffef9d0  
3fff09e0:  3fffdad0 00000000 3ffef9c8 40207008  
3fff09f0:  feefeffe feefeffe 3ffef9e0 40100710  
<<<stack<<<

WilkoV, 2018-07-16:
I have the same problem. It happens all the time after the ESP8266 reconnects after an bad internet connection.

cmtmc, 2019-01-21:
After switching to new version - esp8266 always crashes after some time with Fatal exception 29(StoreProhibitedCause): epc1=0x4021daae, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

Please note that old version works fine.

Old .cpp is 16.623 bytes. New failing .cpp is 23.025 bytes.

cmtmc, 2019-01-21:
It became a little bit clearer. I've been querying bot.getUpdates every 1,000 ms. After increasing to 10,000 ms there was no crash for 20 mins. Bot processed several messages. But then on bot.sendMessage it crashed again with Fatal exception 29(StoreProhibitedCause): epc1=0x4021f926, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

Hope this helps.

Also I have TickerScheduler with 6 - from 3 sec to 3600 - maybe that implies.

Thank you.

array81, 2019-04-30:
I have a similar problem.
If not active the bot client my code works without problem, but when I active it I get the problem.

Any solution?