witnessmenow/Universal-Arduino-Telegram-Bot #87: Problem with ArduinoJson library

BiggyBaron, 2018-08-05:
Dear Developers,
I have faced a problem while just trying to replicate one of the ESP32 examples (Echo bot), it is not compiling because of <ArduinoJson> library, which is installed.
Here is the code:
```#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

// Initialize Wifi connection to the router
char ssid[] = "blablablaba";     // your network SSID (name)
char password[] = "blablablaba"; // your network key

// Initialize Telegram BOT
#define BOTtoken "blablablaba"  // your Bot Token (Get from Botfather)

WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done
bool Start = false;

void handleNewMessages(int numNewMessages) {
  Serial.println("handleNewMessages");
  Serial.println(String(numNewMessages));

  for (int i=0; i<numNewMessages; i++) {
    String chat_id = String(bot.messages[i].chat_id);
    String text = bot.messages[i].text;

    String from_name = bot.messages[i].from_name;
    if (from_name == "") from_name = "Guest";

    if (text == "/send_test_action") {
        bot.sendChatAction(chat_id, "typing");
        delay(4000);
        bot.sendMessage(chat_id, "Did you see the action message?");

        // You can't use own message, just choose from one of bellow

        //typing for text messages
        //upload_photo for photos
        //record_video or upload_video for videos
        //record_audio or upload_audio for audio files
        //upload_document for general files
        //find_location for location data

        //more info here - https://core.telegram.org/bots/api#sendchataction
    }

    if (text == "/start") {
      String welcome = "Welcome to Universal Arduino Telegram Bot library, " + from_name + ".\n";
      welcome += "This is Chat Action Bot example.\n\n";
      welcome += "/send_test_action : to send test chat action message\n";
      bot.sendMessage(chat_id, welcome);
    }
  }
}

void setup() {
  Serial.begin(115200);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

[... 231 lines cut ...]
                 from /Users/naboo/Documents/Arduino/libraries/UniversalTelegramBot/src/UniversalTelegramBot.cpp:36:
/Users/naboo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
/Users/naboo/Documents/Arduino/libraries/UniversalTelegramBot/src/UniversalTelegramBot.cpp:685:25: error: 'jsonBuffer' was not declared in this scope
   JsonObject &payload = jsonBuffer.createObject();
                         ^
/Users/naboo/Documents/Arduino/libraries/UniversalTelegramBot/src/UniversalTelegramBot.cpp:703:57: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::JsonObject'
     JsonObject &replyMarkup = payload.createNestedObject("reply_markup");
                                                         ^
/Users/naboo/Documents/Arduino/libraries/UniversalTelegramBot/src/UniversalTelegramBot.cpp:705:23: error: expected ';' before 'keyboardBuffer'
     DynamicJsonBuffer keyboardBuffer;
                       ^
/Users/naboo/Documents/Arduino/libraries/UniversalTelegramBot/src/UniversalTelegramBot.cpp:706:31: error: 'keyboardBuffer' was not declared in this scope
     replyMarkup["keyboard"] = keyboardBuffer.parseArray(keyboard);
                               ^
Несколько библиотек найдено для "WiFi.h"
 Используется: /Users/naboo/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
Не используется: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Ошибка компиляции для платы "WeMos" WiFi&Bluetooth Battery.
```

Compiled by ArduinoISP with esp-idf installed (other codes works perfectly).  
Compiled on Mac OS x 13+.

witnessmenow, 2018-08-05:
> examples (Echo bot), it is not compiling because of library, which is

Downgrade the arduino json library to 5.x through the library manager,
there was breaking changes in 6.x that's I haven't got a chance to fix yet

Brian

BiggyBaron, 2018-08-05:
@witnessmenow Thank you for your fast response, it worked for me!

closed 2018-08-05 by BiggyBaron

Mrtian2, 2018-12-11:
> Dear Developers,
> I have faced a problem while just trying to replicate one of the ESP32 examples (Echo bot), it is not compiling because of library, which is installed.
> Here is the code:
>
> ``` [...]

sorry can you tell me which arduino json you using i tried all 5.x   it shown 
Arduino: 1.8.7 (Windows 10), Board: "Arduino Nano, ATmega328P"

Build options changed, rebuilding all
C:\Users\acer99\AppData\Local\Temp\ccuDHj48.ltrans0.ltrans.o: In function `main':

C:\Users\acer99\Desktop\yeqiang\Arduino\hardware\arduino\avr\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Users\acer99\Desktop\yeqiang\Arduino\hardware\arduino\avr\cores\arduino/main.cpp:46: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

witnessmenow, 2018-12-11:
You have your board set to nano, this library doesn't run on a nano