witnessmenow/Universal-Arduino-Telegram-Bot #94: Allways returns new message then crashes afrer

MichaelGaylor, 2018-10-30:


MichaelGaylor, 2018-10-30:
Hi this is firstly a great library but I need some help please im pulling my hair out. I can send a message but not receive any... her is the code and the serial responses. ( it is connected fine and Im using the SIM800L with the MEGA Arduino. I send messages but it just ignores them ? Help...…...

// Select your modem: #define TINY_GSM_MODEM_SIM800 // #define TINY_GSM_MODEM_SIM808 // #define TINY_GSM_MODEM_SIM900 // #define TINY_GSM_MODEM_A6 // #define TINY_GSM_MODEM_A7 // #define TINY_GSM_MODEM_M590
#define TINY_GSM_MODEM_SIM800
#include <TinyGsmClient.h>
//#include <ArduinoHttpClient.h>
#include <UniversalTelegramBot.h>

#define Serial Serial
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(7, 8); // RX, TX
#define SerialAT Serial1

#include <Arduino.h>
#include <StreamDebugger.h>

StreamDebugger debugger(SerialAT, Serial);

TinyGsm modem(debugger);

//TinyGsm modem(SerialAT);

const char apn[]  = "payandgo.o2.co.uk";   //apn
const char user[] = "payandgo";   //username
const char pass[] = "password";   //pass

// Initialize Telegram BOT
#define BOTtoken "xxxxxxxxx:xxxxxxxxxxxxxxxxxxxxxxxxx"  // your Bot Token (Get from Botfather)

TinyGsmClientSecure client(modem);
//TinyGsmClient client(modem);

UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done
bool Start = false;
const int ledPin = 13;
int ledStatus = 0;
enum ledModes {ON = 0, OFF, BLINK, PWM}; // define modes of operation for LED (inverse polarity)
enum ledModes ledmode = OFF; //  Start with off

void setup() {
  //bot.longPoll = 60;
  Serial.begin(9600);
  delay(10);

  // Set GSM module baud rate

 SerialAT.begin(9600);
bot._debug=true; // uncomment to see debug messages from bot library
  delay(3000);

   Serial.println(F("Initializing modem..."));
    modem.restart();
    delay(5000);

  String modemInfo = modem.getModemInfo();

 Serial.print(F("Modem: "));

 Serial.println(modemInfo);

  // Unlock your SIM card with a PIN

  //modem.simUnlock("0000");

  modem.sendAT(GF("+CIPSSL=1"));    //enable ssl. maybe don't need
  modem.waitResponse();

[... 319 lines cut ...]
>
 
DATA ACCEPT:1,2
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","149.154.167.220","443","CONNECTED"

OK
AT+CIPRXGET=4,1

+CIPRXGET: 1,1

+CIPRXGET: 4,1,407

OK
AT+CIPRXGET=2,1,63

1, CLOSED

ERROR

u-l-m-i, 2018-10-31:
I'm interested in an solution too, as it looks I'm struggling with the same problem. My setup is based on an SIM800 model & ESP8266 using TinyGSM.

blokerun, 2023-01-30:
fbi1671, can i have your complete code?