witnessmenow/Universal-Arduino-Telegram-Bot #103: Autentification for Telegram Bot

burdeyniyua, 2019-01-09:
Hello! I want to do something like smart home with telegram-bot and esp8266.
I want to do autentification for me and my wife only. 
I looked for [https://github.com/witnessmenow/Immersion-Bot/blob/master/Immersion-Bot.ino](url) ? but I can't compile it - `'numberAuthUsers' was not declared in this scope`.
Can any body help? How I can flash led [https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/FlashLED/FlashLED.ino](url) with autentification by userID?

witnessmenow, 2019-01-09:
Hi, the immersion bot was built using an old version of the library and I
guess doesn't work as it is anymore.

To add what your talking about just check that the chat id of the recieved
message matches your or your wife's in the recieved message method, and
don't respond to the message if it doesn't match

bot.messages[i].chat_id

burdeyniyua, 2019-01-09:
It must be something like this? 
```
if (bot.messages[i].chat_id == "myID" || "wifeID") {
......program code.... body-bot
}
if (bot.messages[i].chat_id != "myID" || "wifeID") {
bot.sendMessage(chat_id, "You are not Administrator of this bot", "");
}
```

burdeyniyua, 2019-04-04:
You can close issue. Cod works.