witnessmenow/Universal-Arduino-Telegram-Bot #104: The Bot doesn't receives anything

DaveCalaway, 2019-01-13:
Hi every body,
I have decide to use again the "ledbot" i edited one years ago over my esp8266. 
I loaded the sketch, crated a new bot with bot father, and added in to the sketch the token and the WiFi info. 
Over the serial, the esp8266, is connected. 

Now, when I try to talk with the bot, over the chat, on the /start command, nothing happens. 

I download again the library, opened a new bot, but nothing change. 
I have use 3 different esp8266, other examples in the library example, but still at the same point.

Am I the only one with this problem?

witnessmenow, 2019-01-13:
Try changing your esp8266 core to 2.4

Tool->board->board manager

Search for esp8266 and select 2.4 from the drop down and click save/update

mnemocron, 2019-01-13:
I just had the same issue with any Telegram Bot library that I tried on my NodeMCU.
**I downgraded the ESP Core to 2.4.2 - now it works!**
Yesterday I had a similar issue with the ArduinoJson Library on the beta channel. 
This "upgrade libraries" feature in Arduino is somewhat misleading.

DaveCalaway, 2019-01-13:
Right, the last esp8266 Arduino Core is 2.5.0**beta2** .  
The last stable is the 2.4.2.  
Some for the last version of the Json Arduino, the last rev is a beta, don't use the Beta but the Stable.

edugargon, 2019-01-18:
thank you!
all works fine with 2.4.2

MrTilllus, 2019-03-10:
can you maybe upload your sketch because when i try to connect with my sketch from 2.5 to my wifi with core 2.4.2 it does not

witnessmenow, 2019-03-10:
On 2.5 add client.setInsecure(); to the setup of the sketch

mrsufgi, 2019-03-13:
@witnessmenow perhaps we need to add this the the docs. 

had the same issues (with ALL telegram bots 🗡 ), downgrading to 2.4.2 works great!

juanmcasillas, 2019-04-21:
Hello.

I have the following configuration:

esp8266 by ESP8266 version 2.5.0
I can't connect with api.telegram.org.
I set the debug to true, and the problem was in the connection.

new core uses BEARSSL instead axTLS so the semantic changes a little bit.
You need to accept all the connections in SSL, without validating the cert.
So you only have to put this:

` client.setInsecure();`

My setup:

`void setup() {
  bot._debug = true;
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  client.setInsecure();
}`

TheUserDead, 2019-08-09:
And if you get nothing when try these advices, check webhook is disabled for your bot. I got angry with myself when I realized this

sindar225, 2020-02-05:
I'm using platformio IDE and trying to make it working on Wemos D1 mini and none of the proposed solutions did work for me. I tried to set client.setInsecure(), use core 2.4.2 or switch to WiFiClientSecureAxTLS - no luck. But what did work for me is changing server's IP in UniversalTelegramBot.h. I just some lines of code from https://github.com/fedrux/ESP8266-TelegramBot and it did the trick. Here's my setup:

**platformio.ini**
```
[env:d1_mini]
platform = espressif8266
board = d1
framework = arduino
upload_port = /dev/cu.wchusbserial1420

lib_deps =
   # Using a library name
   ESP8266-TelegramBot=https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot.git#V1.2.0
```
**UniversalTelegramBot.h**
```
#define HOST "149.154.167.198"
#define SSL_PORT 443
#define HANDLE_MESSAGES 1

//unmark following line to enable debug mode
#define _debug

typedef bool (*MoreDataAvailable)();
typedef byte (*GetNextByte)();
typedef byte* (*GetNextBuffer)();
typedef int (GetNextBufferLen)();

const char* fingerprint = "37:21:36:77:50:57:F3:C9:28:D0:F7:FA:4C:05:35:7F:60:C1:20:44";  //Telegram.org Certificate 
```

javierferwolf, 2020-03-01:
hi @sindar225  I have the same problem,  I have tried several things without result, I use core 2.4.2 .  could you please explain in more detail what steps you followed,  because I've done what you've described but it doesn't work.   or could you please send me the .h .cpp files? because something i'm not doing right!
thank you

witnessmenow, 2020-03-01:
Core 2.4 does not work anymore, telegram require a TLS level core 2.4
doesn't support.

Upgrade to 2.5 or higher and add "client.setInsecure();" anywhere in the
setup of your sketch

haorfani, 2020-03-23:
Hi,

In the following sketch I am trying to send a telegram message through the bot every 5 seconds and it doesn't work. I am able to connect to wifi but I am not able to send messages to telegram successfully. I followed all the instructions to set up the bot and got the token and chatid. What am I am missing here?

``` /*******************************************************************
 *  Push Notificaiton Demo                                         *
 *  Using IFTTT and Telegram to send alerts to your phone          *
 *                                                                 *
 *  Telegram Library:                                              *
 *  https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot *
 *                                                                 *
 *  IFTTT Maker Library:                                           *
 *  https://github.com/witnessmenow/arduino-ifttt-maker            *
 *                                                                 *
 *  By Brian Lough                                                 *
 *******************************************************************/

//Including the two libraries
#include <UniversalTelegramBot.h>

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

//------- WiFi Settings -------
char ssid[] = "XXXXXXXXX";       // your network SSID (name)
char password[] = "YYYYYYY";  // your network key

// ------- Telegram config --------
#define BOT_TOKEN "966974293:AAEU7UOno-Cn1bwA9t276PwYFaLfyir8oeY"  // your Bot Token (Get from Botfather)
#define CHAT_ID "-1094994126" // Chat ID of where you want the message to go (You can use MyIdBot to get the chat ID)

// SSL client needed for both libraries
WiFiClientSecure client;

UniversalTelegramBot bot(BOT_TOKEN, client);

String ipAddress = "";

volatile bool telegramButtonPressedFlag = false;
volatile bool iftttButtonPressedFlag = false;

void setup() {

  Serial.begin(115200);

  client.setInsecure();

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);

  ipAddress = ip.toString();

}

void sendTelegramMessage() {
  String message = "SSID:  ";
  message.concat(ssid);
  message.concat("\n");
  message.concat("IP: ");
  message.concat(ipAddress);
  message.concat("\n");
  if(bot.sendMessage(CHAT_ID, message, "Markdown")){
    Serial.println("TELEGRAM Successfully sent");
  }
  Serial.println(message);
  }

void loop() {
     
    sendTelegramMessage();
    delay(5000);
    
}```

zenercurrent, 2020-04-02:
hey @haorfani, is your esp8266 core still v2.4.2? Telegram had an update so you have to update to v2.5

sefasenlik, 2020-05-28:
> On 2.5 add client.setInsecure(); to the setup of the sketch
> […](#)

It works!