witnessmenow/Universal-Arduino-Telegram-Bot #108: Send float variable

burdeyniyua, 2019-02-17:
Hello! I have a problem to send float variable to telegram.
```

                        if (text == "temp") {//Команда температуры
                          tmp = bme.readTemperature();
                        String temperatura = "Temp=";
                        temperatura += tmp;
                        temperatura +=  "*C\n";
                       // temperatura += "Pressure="+ bme.readPressure()+ "Pa\n\n";
                        bot.sendMessage(chat_id, temperatura, "Markdown");
                        }
```

This code did not working(

burdeyniyua, 2019-02-17:
```
  tmp = bme.readTemperature();
  bot.sendMessage(chat_id, "Temp=", tmp& , "");
```

How I can send message with my variable?
When I send message to serial port - I have my float variable.

```
                            Serial.print("Temperature = ");
                            Serial.print(tmp);
                            Serial.println(" *C");
```

witnessmenow, 2019-02-17:
Try

temperatura += String(tmp);

MIKHANYA, 2021-09-24:
I am trying to write my own bot too, and faced the problem of passing the value of a variable to the messenger, is there a complete clear example? This code question is written in scraps and is incomprehensible

witnessmenow, 2021-09-24:
Look up how to convert a int/float to a String

int value = 44;
String strValue = String(value);

Now you can send strValue

closed 2021-09-24 by witnessmenow