witnessmenow/Universal-Arduino-Telegram-Bot #109: [question] updating message

leonardpitzu, 2019-02-25:
is there and api that you can use to update the same message in a loop? think a thermometer where the user makes a request and the "arduino" keeps pushing an updated value until a "stop" command is received.

witnessmenow, 2019-02-25:
The telegram bot API supports editing of messages, but it's not currently a
feature of this library. It is something I'd like to add but I'm not sure
when I'll get time

Edit message is the endpoint, which is really similar to just sending a
message

https://core.telegram.org/bots/api#editmessagetext

But the biggest change would be that the library doesn't currently do
anything with the response to sending a message, and it would need to to
get the id of the message to edit

witnessmenow, 2019-02-25:
Actually I think I completely missunderstood the question. Just set a
Boolean flag in your sketch that if it's set your loop does something with,
that is set to false when you send the /stop command

leonardpitzu, 2019-02-25:
The first answer was the one i was looking for; the second one I know :-) I patched you lib for getting back the message_id and I try to edit the message but I get back some answer that the message can or be edited. I will call it a night as I’m in the office for 12h and my patching work is worst than usual.

RomeHein, 2020-04-07:
Hey, I’ve put together a pull request that add (mostly) that feature:

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/pull/155

@leonardpitzu, could you confirm that’s what your were looking for?

fritsjan, 2020-07-28:
I wanted the same and took #155 as a guide to make a working pull-request from the latest master, tested and I also wrote a working example to show how people can use it, see #190