witnessmenow/Universal-Arduino-Telegram-Bot #112: Parsing messages

burdeyniyua, 2019-04-04:
Hello word! I want to do timer.

burdeyniyua, 2019-04-04:
I want to send message **timer 1** for enable timer to one minute.
**timer 5** to enable timer too 5 minutes.
**Can I use any masks in `text`?
I want to send key word _timer_  and argument for it.**

```
if (text == "timer &int") {
//parsing timer and argument                       
                                    }
```

not interesting write commands to all arguments of timer (1-1440) 
```
if (text == "timer 1") {
//enable timer for 1 minute                      
                                    }
```
..........
........
```
if (text == "timer 1440") {
//enable timer for 1440 minute                      
                                    }
```

witnessmenow, 2019-04-04:
Hey,

I do a similar thing in a project here

https://github.com/witnessmenow/Simple-Home-Automation-With-Telegram/blob/master/LedControl/LedControl.ino#L100

to adjust it to your case:
`if (text.startsWith("timer")) {
        text.replace("timer ", ""); //This will remove "timer " from your text, leaving you with just the number
        int timeRequested = text.toInt(); // This will return the number
}`

burdeyniyua, 2019-04-05:
Great! It's realy cool to send arguments to telegram bot! Thanks! It's wokrs! 
You can close this Issue.