witnessmenow/Universal-Arduino-Telegram-Bot #118: Connection problem with ESP8266 2.5.0 library and Arduino IDE 1.8.9

array81, 2019-04-30:
If I try to compile my code with ESP8266 2.5.0 (stable) library and Arduino IDE 1.8.9 I get problem on connection to server.

GET Update Messages
[BOT]Connecting to server
[BOT]Conection error
Received empty string in response!

I just have try to use HTTPSRedirect, the result is the same.

rkulyn, 2019-05-31:
Has the same issue. I used debug mode and found WiFiClientSecure client failed to connect. I tried to replace it with unsecure and had some response from telegram nginx. So issue is nearby this library. Currently I still don't have working solution

witnessmenow, 2019-05-31:
Use Wificlientsecure and set client.setInescure(); in your setup

rkulyn, 2019-05-31:
> Use Wificlientsecure and set client.setInescure(); in your setup
> […](#)

Still not working
'class BearSSL::WiFiClientSecure' has no member named 'setInescure'

witnessmenow, 2019-06-01:
setInsecure() ,sorry I spelled it wrong

sanderkob, 2019-08-30:
I am using esp8266 core 2.5.2. It works well by forcing the use of the (deprecated) axTLS client, not the current BearSSL client. With axTLS the performance for core 2.5.2 is similar to that of core 2.4.2 (more heap and faster than with BearSSL). And no need for client.setinsecure() by the way. 
See https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/HTTPSRequestAxTLS/HTTPSRequestAxTLS.ino

consolacion, 2020-01-05:
> client.setInsecure();

Thank you. Suddenly it is all working

r-ryabyy, 2020-01-19:
I have used the suggestion from @sanderkob - using axTLS, but probably better to update the code with newer libraries, since it seems this one can be deprecated and disbled soon, right ?

CyborgAndy, 2020-02-07:
Thank you @sanderkob

sanderkob, 2020-02-07:
It appears my workaround using (deprecated) axTLS does not work anymore, perhaps Telegram ended TLS 1.1 support? Anyway, I am now running succesfully with BearSSL and client.setInsecure()

CyborgAndy, 2020-02-10:
Yes, you are right. Last week your workaround is still worked. But it is no longer working. 
Now I am using BearSSL and client.setFingerprint(). I get telegram api site fingerprint from https://www.grc.com/fingerprints.htm

array81, 2020-02-13:
I cannot understand how get the fingerprint.

CyborgAndy, 2020-02-13:
On the site https://www.grc.com/fingerprints.htm in section "Custom Site Fingerprinting" enter api.telegram.org in textbox and press button "Fingerprint Site". You will get "Security Certificate's Authentic Fingerprint" such as "BB:DC:45:2A:07:E3:4A:71:33:40:32:DA:BE:81:F7:72:6F:4A:2B:6B".
Then paste this fingerprint as uint_8 array in your skecth
const uint8_t fingerprint[20] = { 0xBB, 0xDC, 0x45, 0x2A, 0x07, 0xE3, 0x4A, 0x71, 0x33, 0x40, 0x32, 0xDA, 0xBE, 0x81, 0xF7, 0x72, 0x6F, 0x4A, 0x2B, 0x6B };
Use client.setFingerprint(fingerprint); in setup block.

javierferwolf, 2020-02-24:
hi @CyborgAndy could you please write an example of how to use the library with what you wrote before..please?

morrowwm, 2020-02-28:
Thank you for the solution CyborgAndy. Adding those two lines:

```
const uint8_t fingerprint[20] = { 0xBB, 0xDC, 0x45, 0x2A, 0x07, 0xE3, 0x4A, 0x71, 0x33, 0x40, 0x32, 0xDA, 0xBE, 0x81, 0xF7, 0x72, 0x6F, 0x4A, 0x2B, 0x6B };
...
client.setFingerprint(fingerprint); in setup block.
```
has my bot working again.

javierferwolf, 2020-02-28:
hi @morrowwm  and in relation to the client parameter    "UniversalTelegramBot bot(BOTtoken, client); "  What should be done?  change "client" to fingerprint?  and WiFiClientSecure client;   to  WiFiClientSecure fingerprint;     
sorry for my ignorance but I don't know how to add this solution!

morrowwm, 2020-02-28:
You might post the code you are trying to fix. In general though: put this line

`const uint8_t fingerprint[20] = { 0xBB, 0xDC, 0x45, 0x2A, 0x07, 0xE3, 0x4A, 0x71, 0x33, 0x40, 0x32, 0xDA, 0xBE, 0x81, 0xF7, 0x72, 0x6F, 0x4A, 0x2B, 0x6B };`

somewhere in the preamble of your sketch, and the second:
`client.setFingerprint(fingerprint);`
near the bottom of the setup section

javierferwolf, 2020-02-29:
@morrowwm hi thank you for your answer ther code simple

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

// Initialize Wifi connection to the router
char ssid[] = "xxxxx";     // your network SSID (name)
char password[] = "xxxxx"; // your network key

// Initialize Telegram BOT
#define BOTtoken "xxxxxx"  // your Bot Token (Get from Botfather)

WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done
bool Start = false;

const int ledPin = 13;
int ledStatus = 0;

void handleNewMessages(int numNewMessages) {
  Serial.println("handleNewMessages");
  Serial.println(String(numNewMessages));

  for (int i=0; i<numNewMessages; i++) {
    String chat_id = String(bot.messages[i].chat_id);
    String text = bot.messages[i].text;

    String from_name = bot.messages[i].from_name;
    if (from_name == "") from_name = "Guest";

    if (text == "/ledon") {
      digitalWrite(ledPin, HIGH);   // turn the LED on (HIGH is the voltage level)
      ledStatus = 1;
      bot.sendMessage(chat_id, "Led is ON", "");
    }

    if (text == "/ledoff") {
      ledStatus = 0;
      digitalWrite(ledPin, LOW);    // turn the LED off (LOW is the voltage level)
      bot.sendMessage(chat_id, "Led is OFF", "");
    }

    if (text == "/status") {
      if(ledStatus){
        bot.sendMessage(chat_id, "Led is ON", "");
      } else {
        bot.sendMessage(chat_id, "Led is OFF", "");
      }
    }

    if (text == "/start") {
      String welcome = "Welcome to Universal Arduino Telegram Bot library, " + from_name + ".\n";
      welcome += "This is Flash Led Bot example.\n\n";
      welcome += "/ledon : to switch the Led ON\n";
      welcome += "/ledoff : to switch the Led OFF\n";
      welcome += "/status : Returns current status of LED\n";
      bot.sendMessage(chat_id, welcome, "Markdown");
    }
  }
}

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  client.setInsecure();
  // attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  pinMode(ledPin, OUTPUT); // initialize digital ledPin as an output.
  delay(10);
  digitalWrite(ledPin, LOW); // initialize pin as off
}

void loop() {
  if (millis() > Bot_lasttime + Bot_mtbs)  {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);

    while(numNewMessages) {
      Serial.println("got response");
      handleNewMessages(numNewMessages);
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }

    Bot_lasttime = millis();
  }
}

the error: exit status 1
'class axTLS::WiFiClientSecure' has no member named 'setInsecure'

morrowwm, 2020-03-01:
Here is your code with the two lines added, as a text file. It compiled for me.

[javierferwolf.txt](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/files/4272003/javierferwolf.txt)

javierferwolf, 2020-03-01:
@morrowwm thank you for your help and answer!!  but  forget to mention, that the version of my core  2.4.2 and i get this problem:

 'class axTLS::WiFiClientSecure' has no member named 'setFingerprint'

morrowwm, 2020-03-01:
I am using core 2.5.0