witnessmenow/Universal-Arduino-Telegram-Bot #119: The ESP32 examples  they stopped working

viquiuxd, 2019-05-05:
Hi. I had an application in my "ESP32" that I had created in January 2019. I updated it with a small spelling mistake but when I loaded it it no longer works. I reloaded my old code and it did not work either.

Desperate I went to review the examples for the ESP32 and to try them with a new bot these did not work either. What could I do to fix it?

When I created the application I used IDE 1.8.5 and my "ESP32" has version 1.0.0. At this moment I am using the same versions and the error continues

JuarezValencia, 2019-05-10:
Hi, this problem is generated by esptool.py (2.6.X). If you downgrade it all the examples going to work again.

viquiuxd, 2019-05-10:
Thank you! Now it works

closed 2019-05-10 by viquiuxd