witnessmenow/Universal-Arduino-Telegram-Bot #120: Add message_id to received results

rkulyn, 2019-06-01:
Please add **message_id** key to received messages body.
This field is required to check if duplicate messages are given from telegram.
Steps to reproduce:

1. Run "examples/ESP32/CustomKeyboard/InlineKeyboardMarkup/InlineKeyboardMarkup.ino"
2. Open telegram in iOS mobile application.
3. Click on any menu item in application.
4. See message got in log.
5. Hide telegram application and restore it again.
6. See 2 duplicated messages sent from telegram.

Does not reproduce on web application.

Bot version: 1.1.0
Mobile iOS application 5.7

fritsjan, 2020-07-27:
yes please!

fritsjan, 2020-07-27:
I think this is already available: 
bot.messages[i].update_id

fritsjan, 2020-07-28:
I found out update_id is not the same as message_id.
I build a pull-request which added update_id.
See #190