witnessmenow/Universal-Arduino-Telegram-Bot #124: Problem with DynamicJsonBuffers in ArduinoJson version 6

grudloff, 2019-08-11:
Anyone else has issues with compiling the examples given due to incompatibilities of the library with the newest version of ArduinoJson?

Downgrading to ArduinoJson v5 works fine.

jamesmyatt, 2019-12-07:
Can you rename the V1.2.0 branch to "develop" or similar, please? Otherwise, people won't get updated when master overtakes it.

Anton-V-K, 2020-10-21:
I wonder why V1.2.0+ isn't still available in the Library Manager of Arduino IDE - only V1.1.0 is available there.

witnessmenow, 2020-11-08:
This issue will be fixed when the new version is released

> 
> 
> I wonder why V1.2.0+ isn't still available in the Library Manager of Arduino IDE - only V1.1.0 can be installed.
> I guess, adding the tag like 'v1.2.0' will do the trick.

Because of #198 and I have a full time job, 2 kids, lots of libraries, a tindie store to run and a youtube channel.

I didn't want to release a library without testing the examples.

closed 2020-11-08 by witnessmenow