witnessmenow/Universal-Arduino-Telegram-Bot #128: Sending photos from SD is successful, but I don't receive them in my Telegram chat.

javi-14, 2019-10-03:
Hello everyone, first of all thanks for this amazing project.

I am having some issue trying to send a photo from an SD card in a ESP32-CAM module (Ai Thinker). I am using the "PhotoFromSD.ino" example, adapted for ESP32 and SD_MMC fuctionality. My SD card works correctly, and inside it contains a few images. The library version that I am using is 1.1.0 .

I have already checked that the image is in the SD card and the name is correct. Even the sent() funcition is successful, because I recieve the confirmation in my serial port. But somehow, I never receive the image in my Telegram chat. I have checked that I can recieve text messages from my bot correctly, so my problem is just with images.

Here is my serial monitor after sending a few messages to the bot. 

> got response
> /picture2.jpg....was successfully sent
> got response
> /picture2.jpg....was successfully sent
> got response
> /picture2.jpg....was successfully sent
> got response
> /picture2.jpg....was successfully sent
> got response
> /picture2.jpg....was successfully sent

And here is the code that I am using (library, definitions and setup are just ok I think).

```
void loop() {

   if (millis() > Bot_lasttime + Bot_mtbs)  {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);

    while(numNewMessages) {
      Serial.println("got response");

      String chat_id = bot.messages[0].chat_id;
      String file_name = "/picture2.jpg";
    
      fs::FS &fs = SD_MMC; 
      File myFile = fs.open(file_name, FILE_WRITE);    

    if (myFile) {
      Serial.print(file_name);
      Serial.print("....");
  
      //Content type for PNG image/png
      String sent = bot.sendPhotoByBinary(chat_id, "image/jpeg", myFile.size(),
          isMoreDataAvailable,
          getNextByte);

      bot.sendMessage(chat_id, "Hi:","");         //THIS WORKS
      bot.sendMessage(chat_id, chat_id,"");       //THIS WORKS
  
      if (sent) {
        Serial.println("was successfully sent");  //THIS IS PRINTED IN MY SERIAL MONITOR
      } else {
        Serial.println("was not sent");
      }
  
      myFile.close();
    } else {
      // if the file didn't open, print an error:
      Serial.println("error opening photo");
    }
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }
    Bot_lasttime = millis();
  }
}

bool isMoreDataAvailable(){
  return myFile.available();
}

byte getNextByte(){
  return myFile.read();
}
```

Thank you all in advance.

ambanmba, 2020-12-24:
Did this ever get resolved? I now have this problem on a piece of code that works perfectly on on ESP32 but not on two others. Exactly the same symptoms and the code is very similar.

javi-14, 2020-12-24:
> Did this ever get resolved? I now have this problem on a piece of code that works perfectly on on ESP32 but not on two others. Exactly the same symptoms and the code is very similar.

Hello,
I have never come back to this, so I have no idea. I didn't continue with the project.

ambanmba, 2020-12-25:
[image]
With debugging switched to maximum you can see there is a problem in ssl_client.cpp. If you send very small images (small icons and such) that are about the same size as a text message they go through. But anything largish (100k) triggers this error.

ambanmba, 2020-12-29:
I was able to solve this problem be replacing the WiFiClientSecure library with the source that’s dated 2-Oct-19. When I upgraded my Mac to Big Sur, I also had to upgrade the ESP32 library - and that one also has a newer WiFiClientSecure which doesn’t work on larger files.

The working version of WiFiClientSecure library is in the esp32-1.0.4 package on GitHub.

javi-14, 2020-12-29:
Thank you, @ambanmba

akil1947, 2023-06-04:
> I was able to solve this problem be replacing the WiFiClientSecure library with the source that’s dated 2-Oct-19. When I upgraded my Mac to Big Sur, I also had to upgrade the ESP32 library - and [...]

thanks! this worked!