witnessmenow/Universal-Arduino-Telegram-Bot #131: Not clear how to use TelegramBot with GPRS SIM800 modem

jbrepogmailcom, 2019-11-14:
Hello,

I do not use WiFi but serial communication with modem to connect to internet. Is it possible to use the UniversalTelegramBot? I did not find any way how.

I do GET connection over serial this way. How can I include Telegram bot like this?

const char server[] = "api.thingspeak.com";
const char resource[] = "/update.json?api_key=TX.........38QPAHK&field1=";

const int  port = 80;                             // server port number

// Set serial for debug console (to Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to SIM800 module)
#define SerialAT Serial1

  TinyGsm modem(SerialAT);

TinyGsmClient client(modem);
HttpClient http(client, server, port);

  // Set serial monitor debugging window baud rate to 115200
  SerialMon.begin(115200);

  SerialMon.print(F("Performing HTTP GET request... "));
  int err = http.get(resource+String(Vcc));
  if (err != 0) {
    SerialMon.println(F("failed to connect"));
    delay(10000);
    return;
  }

  int status = http.responseStatusCode();
  SerialMon.print(F("Response status code: "));
  SerialMon.println(status);
  if (!status) {
    delay(10000);
    return;
  }

  SerialMon.println(F("Response Headers:"));
  while (http.headerAvailable()) {
    String headerName = http.readHeaderName();
    String headerValue = http.readHeaderValue();
    SerialMon.println("    " + headerName + " : " + headerValue);
  }

  int length = http.contentLength();
  if (length >= 0) {
    SerialMon.print(F("Content length is: "));
    SerialMon.println(length);
  }
  if (http.isResponseChunked()) {
    SerialMon.println(F("The response is chunked"));
  }

  String body = http.responseBody();
  SerialMon.println(F("Response:"));
  SerialMon.println(body);

  SerialMon.print(F("Body length is: "));
  SerialMon.println(body.length());

  http.stop();

gruposomel, 2020-08-21:
try it:
https://core.telegram.org/bots/webhooks

witnessmenow, 2020-08-21:
Sim800 does not work anymore because it doesn't support TLS 1.2, and
telegram requires that since earlier this year

closed 2020-08-24 by witnessmenow