witnessmenow/Universal-Arduino-Telegram-Bot #134: esp how use proxy

mohamad-reza, 2020-01-20:
hi i have one problem in my country telegram server is closed and filter
and the esp8266 can not conect to the server of the telegram do you know how can i use the proxy for slove my problem

erfan-azhdari, 2020-07-07:
same problem here. telegram is unfortunately censored in Iran. is there any way to use this library in Iran?

denzen84, 2020-07-07:
Yes, but solution is not easy. Until telegram was blocked in Russia, I used special WiFi hotspot that passes all Internet traffic through Tor service.

witnessmenow, 2020-07-07:
If there is a proxy endpoint for telegram that works you could modify the
library to point to that . (I don't know if that exists)

You should just need to update the telegram_host in UniversalTelegramBot.h

erfan-azhdari, 2020-07-08:
> If there is a proxy endpoint for telegram that works you could modify the library to point to that . (I don't know if that exists) You should just need to update the telegram_host in UniversalTelegramBot.h
> […](#)

Thanks for your reply. I found a proxy endpoint ([This](https://vladcyb-bot.herokuapp.com/) , [github](https://github.com/vladcyb/telegram-bot-mirror)) and it's not blocked (I think they can't block herokuapp.com IPs).
I'll edit the file and see if it works or not.

erfan-azhdari, 2020-07-08:
I edited the file and it works!
(I used my private VPS as the server)
Thank you @witnessmenow.
I think this issue should be closed now.

erfan-azhdari, 2020-07-08:
I uncommented [this](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/e2c901fda41af378bf6fdd6995628bd352430218/src/UniversalTelegramBot.h#L25)(`//#define TELEGRAM_DEBUG 1`) line and got this:

`incoming message length 324`
`Creating DynamicJsonBuffer`
`GetUpdates parsed jsonObj: {"ok":true,"res":{"ok":true,"result":[{"update_id":12345,"message":{"message_id":1,"from":{"id":123456,"is_bot":false,"first_name":"XXXXXXXX","language_code":"en"},"chat":{"id":123456,"first_name":"XXXXXXXX","type":"private"},"date":1594210530,"text":"/start","entities":[{"offset":0,"length":6,"type":"bot_command"}]}}]}}`
`Response contained no 'result'`
`Closing client`
`GET Update Messages`
`[BOT]Connecting to server`
`sending: bot12345:XXXXXXXXXXXXX/getUpdates?offset=1&limit=1`
`Received empty string in response!`
`Closing client`

erfan-azhdari, 2020-07-13:
Any solutions?

witnessmenow, 2020-09-30:
Is this still an issue?