witnessmenow/Universal-Arduino-Telegram-Bot #135: Does not compile with WiFiNINA (Arduino MKR 1010) and ArduinoJson 6.14.1

vaskarni, 2020-01-28:
Hello, 
I am trying to compile for a Arduino MKR 1010 board. Here are my includes.
#include <SPI.h>
//#include <WiFi101.h>
#include <WiFiNINA.h> //Include this instead of WiFi101.h as needed
#include <UniversalTelegramBot.h>

I get the following error:
Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino MKR WiFi 1010"

Build options changed, rebuilding all
C:\Users\Srinivas\Documents\Arduino\libraries\UniversalTelegramBot\src\UniversalTelegramBot.cpp:319:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   DynamicJsonBuffer jsonBuffer;

           ^~~~~~~~~~~~~~~~~~~~~ 

Appreciate your help in fixing this.

witnessmenow, 2020-01-28:
You can either manually upgrade this library to the V1.2 Branch

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/V1.2.0

or it is probably easier to downgrade the ArduinoJson library to 5.X (you can do this through the library manager)

vaskarni, 2020-01-28:
Thank you! I will try that.

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/V1.2.0

closed 2020-09-30 by witnessmenow