witnessmenow/Universal-Arduino-Telegram-Bot #138: Esp8266 fail ssl connection

bonz82, 2020-02-07:
since this wednesday my esp8266 bots dont receive or send telegram mensages.

Any fix for the new ssl conection to telegram?

witnessmenow, 2020-02-07:
see #139 , it's working fine for me. You may need to update your version of ESP8266 core and add `client.setInsecure();` to the setup

[image]

DrXoo, 2020-02-08:
Also checkout the ESP8266 version as #139

array81, 2020-03-28:
Is possible use insecure connection on ESP8266 v.2.4.2 library?
I need use axTLS and not BearSSL so I need use something like setInsecure for axTLS.

witnessmenow, 2020-03-28:
AxTLS doesn't meet the minimum requirements for TLS with the telegram API
as far as I know

Why can't you use bear?

array81, 2020-03-28:
Bear requere more memory of axtls. If I use web server and ssl connetion (for Telegram) with bear on NodeMCU I get out of memory. This is the problem. Do you have idea about It?

witnessmenow, 2020-09-30:
I'm sorry but there isn't really anything I can do about that.

If you use case is quite simple you could try the new Telegram library that I have worked on, it should be a little less memory consuming, but it still requires bear SSL (You can't talk to Telegram without it)

https://github.com/witnessmenow/pinocchio-arduino-telegram-library

closed 2020-09-30 by witnessmenow