witnessmenow/Universal-Arduino-Telegram-Bot #139: The Bot API only accepts incoming TLS 1.2

juangarcia22, 2020-02-07:
since last February 6, the Bot API only accepts incoming TLS 1.2 (or higher) connections. 
Anyone know how to update this library .
This has been published in Bot News  [https://t.me/BotNews]

Thanks.

witnessmenow, 2020-02-07:
Apparently esp8266 supports TLS 1.2 for some time, can you be more specific
to what board you are using and what version of esp8266 core you are using?

juangarcia22, 2020-02-07:
I use esp 8266 d1 mini, IDE arduino board V2.4.2

witnessmenow, 2020-02-07:
[image]
[image]

EchoBot example with `client.setInsecure();` added to the setup working for me. I'm using ESP8266 core 2.5.2

DrXoo, 2020-02-07:
@witnessmenow I tried your workaround but I got an error

**'class axTLS::WiFiClientSecure' has no member named 'setInsecure'** 

Maybe our WifiClients lib versions are different

witnessmenow, 2020-02-07:
Are you using a different WiFi client than the examples? Are you on a
different version of esp8266 to me?

DrXoo, 2020-02-07:
@witnessmenow Indeed we were using different versions.

Now it works flawless, even without the setInsecure. I updated it to 2.6.3 version.

juangarcia22, 2020-02-08:
fantastic!!
I've only had to update the core to 2.6.2 and I've also included client.setInsecure (); in the setup.

Thanks for the help, I've been going around for several days.

juangarcia22, 2020-02-08:
I think this is already solved and can be closed. thanks @witnessmenow.

closed 2020-02-08 by juangarcia22

r-ryabyy, 2020-02-08:
I didn't get how to get it working. Some time ago i've fixed the problem via using:
`#define USING_AXTLS
#include <WiFiClientSecureAxTLS.h>
using namespace axTLS;
#include <UniversalTelegramBot.h>
...

#pragma GCC diagnostic push
#pragma GCC diagnostic ignored  "-Wdeprecated-declarations"
  WiFiClientSecure client;
#pragma GCC diagnostic pop

...`

But since Feb, 6 it stopped working

I'm running ESP8266 on 2.6.3 library

witnessmenow, 2020-02-08:
Why are you using they client over the stranded WifiClientSecure?

r-ryabyy, 2020-02-08:
> Why are you using they client over the stranded WifiClientSecure?
> […](#)

Stated in the examples sources doesn't work anymore.
Which one we can use ? Maybe it's time to update examples ?

r-ryabyy, 2020-02-08:
> @witnessmenow Indeed we were using different versions.
> 
> Now it works flawless, even without the setInsecure. I updated it to 2.6.3 version.

Can you share the way how you are using the library without setInsecure ?

witnessmenow, 2020-02-08:
Either use client.setInsecure(); or set a fingerprint using client.setFingerprint();

https://github.com/witnessmenow/arduino-tindie-telegram-notify/blob/master/tindieTelegramNotify/tindieTelegramNotify.ino#L100

Explained in more detail here: https://youtu.be/HUjFMVOpXBM?t=259

What you were doing previously was the equivalent of using setInsecure.

array81, 2020-02-13:
Where find the fingerprint? Is is the same for all bots or any bot have a fingerprint?