witnessmenow/Universal-Arduino-Telegram-Bot #140: bot not sending any message!

jjojas, 2020-02-09:
The bot is not sending any message to my Chat! any help?

```
#include <UniversalTelegramBot.h>
#include <ESP8266WiFi.h>
#define BOTtoken "xxxxxx"
#include <WiFiClientSecure.h>
#define chat_id "xxxxxx"
#include <ArduinoJson.h>

char ssid[] = "xxxxxx";     // your network SSID (name)
char password[] = "xxxxxx";

const int sensorPin = 7; 
int sensorState = 0;
int lastState = 0;

WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);

void setup() {
  Serial.begin(9600);

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  // attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop()
{ 
  sensorState = digitalRead(sensorPin);
  Serial.println(sensorState);
  bot.sendMessage(chat_id,String(sensorState));

if (sensorState == 1 && lastState == 0) {
  Serial.println("needs water, send notification");
  bot.sendMessage(chat_id,"tes","Siram Tanamanmu!");
  lastState = 1;
  delay(1000);
//send notification
    
  } 
  else if (sensorState == 1 && lastState == 1) {
    //do nothing, has not been watered yet
  Serial.println("has not been watered yet");
  delay(1000);
  }
  else {
    //st
    Serial.println("does not need water");
    bot.sendMessage(chat_id,"tes","TidakButuhAer");
    lastState = 0;
    delay(1000);
  }
  
  delay(100);
}
```

witnessmenow, 2020-02-09:
Update your version of esp8266 core to at least 2.5

Add the following to your setup

client.setInsecure():

jjojas, 2020-02-09:
Where do i put it?

witnessmenow, 2020-02-09:
Your setup()

jjojas, 2020-02-09:
Problem solved! Thanks

Necromancer1982, 2020-02-10:
Hi jjojas,

can you explain, how you solved the problem?

Thanks Andreas

DimChik2903, 2020-02-10:
Hello! I have the task of sending messages in the setup. If I add client.setInsecure(); before sending, nothing works(

Necromancer1982, 2020-02-10:
OK, solved my problem, too.
I've updated the ESP Core to 2.5.0 and inserted
client.setInsecure();
right at the beginning of my setup.

Thank you witnessmenow for the solution!

closed 2020-09-30 by witnessmenow