witnessmenow/Universal-Arduino-Telegram-Bot #143: Longpoll

fmarzocca, 2020-02-13:
My ESP will receive no more than 5 or 6 commands from Telegram in 24 hours, but in such a case, I need a fast response, action and notification.

I have some doubt about using Longpoll. Who can give me Pros & Cons on Longpoll?

witnessmenow, 2020-02-13:
I explain it in this video

https://youtu.be/-IC-Z78aTOs

Basically it's good if the trigger for your code to activate is telegram
and not something external like a sensor or button

fmarzocca, 2020-02-13:
Thank you Brian,
the pro lem is that my application is activated by:

- a telegram command
- a pin interrupt
- a http post

and I am noticing that everything is slowed down (until timeout) if I use longPoll. On the other hand, without a longPoll, my ESP8266 is constantly sending/receiving bits (the blue led is almost always on).

RomanLut, 2020-02-20:
Longpool is implemented using busy wait in this library.
getUpdates() blocks for {longPoll} seconds.
Thus it is not usable in projects where you need to do some tasks in loop().
getUpdates() should just open connection and check if bytes are available in loop() without blocking.
I modified library this way, getUpdates() can be called with any frequency without blocking.
There still will be blocking when bot has to send responce. Or there is connection timeout. To fully fix this, bot has to be based on asynctcp library.

Attached is modified library. 
[esp8266firmware.zip](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/files/4232731/esp8266firmware.zip)

fmarzocca, 2020-02-26:
@RomanLut , do you mean that with your modified library I could set longPoll freely, without blocking other activities in the loop (such as ArduinoOTA.handle() or server.handleClient()?

RomanLut, 2020-02-26:
Do not change longPool memeber, it is private now and set to 50 seconds.
Just call getUpdates() with any frequency without blocking.
There still will be blocking when you send messages.

BTW I have tried to implement bot based on ESPAsyncTCP, but it does not work because TLS 1.2 is not supported by ESPAsyncTCP. api.telegram.org requires TLS 1.2 now.

fmarzocca, 2020-02-27:
I didn't find the private declaration of longpoll in your code, neither its assignment to 50...
Maybe I have a wrong code?

fmarzocca, 2020-02-27:
I got this compiler error:

`undefined reference to `UniversalTelegramBot::sendMessage(String, String, String, bool)'`

RomanLut, 2020-02-27:
Please double check that you use files from archive above. sendMessage() is properly declared():
.h 
bool sendMessage(String chat_id, String text, String parse_mode = "", bool disable_notification = false);
.cpp:
bool UniversalTelegramBot::sendSimpleMessage(String chat_id, String text, String parse_mode, bool disable_notification)

fmarzocca, 2020-02-27:
Yes, I have looked at your code and I have seen it correctly declares the function.
The problem is that the compiler looks like still using the old library! I have replaced .cpp and .h files with yours in <sketchfolder>/libraries/UniversalTelegramBot/src

fmarzocca, 2020-02-27:
Fixed. I had to refresh the libry cache

fmarzocca, 2020-02-27:
Tested.
THIS IS GREAT! Finally I have get rid of all the huge network traffic, and the longPoll is not blocking my other functions in the loop!

Thank you very much, you should issue a Pull Request or fork this library with yours!
Great job, @RomanLut !

fmarzocca, 2020-02-27:
What is the use of disable_notification in sendMessage()?

RomanLut, 2020-02-27:
You can pass true to disable notification sound.
I am sending log messages silently and alarm messages with notification sound, which I set to alarm sound in chat options in telegram app on phone.

fmarzocca, 2020-02-27:
> You can pass true to disable notification sound.
> I am sending log messages silently and alarm messages with notification sound, which I set to alarm sound in chat options in telegram app on phone.

Thanks

RomanLut, 2020-02-27:
Note that this update does not solve all problems. 
If you call sendMessage(), it will block for <5 seconds  and next getUpdates() will block to <5 seconds once.
Worse, if you have connection problems, each getUpdates() will block for 5-50 seconds because client->connect() will block while trying to establish connection to api.telegram.org.
Also note issue https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/38
If you send a message with long text (1300 bytes), library will fall into endless update cycle generating huge traffic. There is a fix which is not  merged.

fmarzocca, 2020-02-27:
Ok, I can accept that short block (<5sec) for my purposes. I hope not to have conncetion problms (the extender AP is close to the device) and I don't need to send long messages.
How can I subscribe or be notified of library's updates?

RomanLut, 2020-02-27:
No, I do not have repo and there will be no more updates from me because I will not be using this library (I can not accept any blocking in my device).

fmarzocca, 2020-02-27:
> No, I do not have repo and there will be no more updates from me because I will not be using this library (I can not accept any blocking in my device).

Maybe Brian @witnessmenow can include your mods into the official library...
There is no other solution to avoid a short block, unless remove the longPoll and accept 30/40 GB of junk network activity per month.
I had a very good experience with Raspberry and library telepot, but that's another story.

fmarzocca, 2020-02-29:
@RomanLut I noticed a little delay in responses (about 4-5 seconds). Is it normal or there is a way to expedite it?

fmarzocca, 2020-02-29:
The small blocking issue in sendMessage is due to checkForOkResponse. Is it strictly necessary?

RomanLut, 2020-02-29:
Library maintains single connection because esp8266 does not have enough memory for two secured connections.
If you call sendMessage(), library closes connection opened for longpool getUpdates(), reopens connection and sends message. Secure connection handshaking requires ~2-5 seconds on esp8266.  I do no think something can be improved here.

fmarzocca, 2020-02-29:
Ok. Could ESP32 be an option to keep 2 connections alive?

RomanLut, 2020-02-29:
I do not have experience with ESP32.

fmarzocca, 2020-03-31:
@RomanLut It's a month that I am using your library in my device. Once a week it happens that the device gets blocked: it is pingable, but it does not respond to Telegram or send messages. I have tried with FW 2.6.1 and 2.5.3, but it's the same. I have to recycle power. Do you have any idea?

fmarzocca, 2020-06-07:
@RomanLut since today Telegram servers has changed something and the library is not working anymore.  I have seen that Brian release v.1.20. Do you have modified your code too? Thanks

witnessmenow, 2020-09-30:
I'm not sure what the current status of this is, is it still worth looking into?

fmarzocca, 2020-10-01:
Yes, I believe it is. I am currently working with Roman's patch.

fmarzocca, 2021-01-01:
> I'm not sure what the current status of this is, is it still worth looking into?

@witnessmenow Brian, is there a chance to embed this mod into the latest library update?

droidprova, 2021-02-07:
that would be the best, as the romanLut patch doesn't work with the latest release 1.3.0. Thank you